Asuswrt-Merlin/asuswrt-merlin.legacy #1444: Orange TV Spain

LoveOFF, 2017-08-07:
Hi,

I have RT-AC66U router with de last firmware version.

For IPTV channels:
The decoder sends an igmp v2 to join a multicast group, the livebox router does igmp proxy and resends that request with origin ip 0.0.0.0 by VLAN 838.

From that moment multicast traffic is received by the VLAN 838 that I must forward the decoder.

I don't know how config it.... i have the config for mikrokit RouterOS:

**/interface vlan
add interface=ether1-gateway name=vlan832 vlan-id=832
add interface=ether1-gateway name=vlan838 vlan-id=838

/ip pool
add name=pool-red-200 ranges=192.168.200.10-192.168.200.100

/ip dhcp-server
add address-pool=pool-red-200 disabled=no interface=ether5-master-local lease-time=23h59m59s name=server-red-200

/ip dhcp-client
add default-route-distance=0 dhcp-options=hostname,clientid disabled=no interface=vlan832

/interface list member
add interface=vlan838 list=orange_tv
add interface=ether5-master-local list=orange_tv

/ip address
add address=192.168.200.1/24 comment="Red STB TV" interface=ether5-master-local network=192.168.200.0
add address=10.20.30.40 interface=vlan838 network=10.20.30.40

/ip dhcp-server network
add address=192.168.200.0/24 dns-server=62.36.225.150,62.37.228.20 gateway=192.168.200.1

/ip firewall filter
add action=accept chain=input in-interface-list=orange_tv log=yes log-prefix=IGMP protocol=igmp
add action=accept chain=forward dst-port=1234,1236 in-interface=vlan838 log=yes log-prefix="TV MULTICAST" out-interface=ether5-master-local protocol=udp src-address=92.58.250.160 src-port=1234
add action=accept chain=input dst-port=1234 in-interface=vlan838 protocol=udp src-address=92.58.250.160 src-port=1234

/ip firewall nat
add action=masquerade chain=srcnat comment="default configuration" out-interface=vlan832

/ip firewall mangle
add action=mark-packet chain=forward comment="PAQUETES TV" dst-port=1234,1236 in-interface=vlan838 new-packet-mark=tv passthrough=yes protocol=udp src-address=92.58.250.160 src-port=1234

/routing igmp-proxy interface
add alternative-subnets=0.0.0.0/0 interface=vlan838 upstream=yes
add alternative-subnets=0.0.0.0/0

/routing igmp-proxy
set query-interval=1m query-response-interval=1m quick-leave=yes**

Can you tell me how config my ASUS? Please.

Thanks very much

LoveOFF, 2017-08-09:
Can someone help me?

RMerl, 2017-08-10:
Please use the support forums if you need help.

closed 2017-08-10 by RMerl