Asuswrt-Merlin/asuswrt-merlin.legacy #1448: Proprietary drivers free version

Bitim, 2017-08-10:
I saw on your website that you are using proprietary drivers for some features:
> enhanced NTFS performance (through the proprietary drivers used by Asus from either Paragon or Tuxera)

Is there any version free of this proprietary drivers?

If not, are there any other parts in released binaries that are closed source? If yes, what are they?

Thanks.

RMerl, 2017-08-10:
It's impossible to compile a version that's 100% free of proprietary code.  The wireless driver and everything associated to it are closed source for instance.

Bitim, 2017-08-13:
> It's impossible to compile a version that's 100% free of proprietary code. The wireless driver and everything associated to it are closed source for instance.

So it won't be really usable. Other third-party firmwares like DD-WRT are also using  these proprietary drivers?

RMerl, 2017-08-13:
OpenWRT (and variations like Lede) is the only one I know of that is 100% sourced.  Blame it on SoC manufacturers like Broadcom and Qualcomm who refuse to open source their drivers.

Bitim, 2017-08-14:
Which not support a lot of Asus's products, reason mentioned in OpenWRT website is:
>     Unsupported due to ASUS' proprietary wireless AC implementation

Thanks!

closed 2017-08-14 by Bitim