Asuswrt-Merlin/asuswrt-merlin.legacy #1449: tproxy kernel support for AC-66U

zw963, 2017-08-13:
I try to resolve one issue, i need use tproxy module support to forward udp.

Please see issue: https://github.com/shadowsocks/shadowsocks-libev/issues/1632

Thanks

zw963, 2017-08-13:
Following is copy from that issue:
```sh
# FREEWEB is china whitelist domain name.
 ipset -N FREEWEB iphash

modprobe xt_TPROXY

$iptables -t mangle -N SHADOWSOCKS
# iptables -t mangle -N SHADOWSOCKS_MARK

$iptables -t mangle -A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN
$iptables -t mangle -A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN

ip rule add fwmark 0x01/0x01 table 100
ip route add local 0.0.0.0/0 dev lo table 100

$iptables -t mangle -A SHADOWSOCKS -p udp -m set --match-set FREEWEB dst -j RETURN

$iptables -t mangle -A SHADOWSOCKS -p udp -j TPROXY --on-port 1082 --tproxy-mark 0x01/0x01
# $iptables -t mangle -A SHADOWSOCKS_MARK -p udp -m set --match-set gfwlist dst -j MARK --set-mark 1

$iptables -t mangle -A PREROUTING -p udp -j SHADOWSOCKS
# $iptables -t mangle -A OUTPUT -j SHADOWSOCKS_MARK

```
But, following error is happen.

```sh
# modprobe xt_TPROXY
modprobe: module xt_TPROXY not found in modules.dep
```
```sh
# ip rule add fwmark 0x01/0x01 table 100
Error: argument "0x01/0x01" is wrong: fwmark value is invalid
```

Is there some clue about how to fix those two issue?

Thanks

zw963, 2017-08-13:
@RMerl , I think i create a similar issue before: https://github.com/RMerl/asuswrt-merlin/issues/982

zw963, 2017-08-13:
following is some describe for this module:

```sh
Transparent proxy support
=========================

This feature adds Linux 2.2-like transparent proxy support to current kernels.
To use it, enable the socket match and the TPROXY target in your kernel config.
You will need policy routing too, so be sure to enable that as well.
```

https://www.kernel.org/doc/Documentation/networking/tproxy.txt

zw963, 2017-08-13:
> To use it, enable NETFILTER_TPROXY, the socket match and the TPROXY target in
your kernel config. 

Please help me, how to do that for AC66U ?

 Firmware:380.67

zw963, 2017-08-14:
Hi, I confirm from the shadowsocks maintainer

```
TPROXY kernel module is required
```

Please see this

RMerl, 2017-08-14:
The RT-AC66U kernel is too old for this option.  It was only added with 2,.6.28 - RT-AC66U is based on 2.6.22.19.

zw963, 2017-08-14:
@RMerl ,  That is to say, if i buy a more new router, this module is worked, right?

Please help give me a support device name which support this feature? 
thanks

zw963, 2017-08-14:
@RMerl , AC68U or AC86U is support that? thanks

zw963, 2017-08-14:
for better support asuswrt-merlin, is there exist a better candidate?

thanks

RMerl, 2017-08-14:
The option exists in the kernel used by ARM-based models such as the RT-AC68U, however I would first need to ensure that the option can be enabled.  Some kernel options cannot be changed, as it modifies the kernel in a way that prevents pre-compiled modules from Tuxera/Trend Micro/Asus from working.

zw963, 2017-08-14:
@RMerl , Thanks to know ARM-based router all support that.

If you ensure about can enable it in AC-68u or whatever, please let me know,
I will buy that for this new feature. 

Thanks

RMerl, 2017-08-16:
Module was added with commit 362ef4f92258e50c99ef175bf92c0ba809e08463, however it's untested (I only made sure it didn't create symbol mismatch errors with existing prebuilt modules).

odkrys, 2017-08-16:
it works well with shadowsocks-libev (ss-redir, transparent proxy) 😄

RMerl, 2017-08-16:
I noticed that the module was already present for router models with WTFast (AC88/AC3100/AC5300).  I've revised the original commit to just extend it to other ARM models rather than modify the default options in config_base.6a.

odkrys, 2017-08-16:
I was using wtfast option to use tproxy module for ac1900p.
Now I don't need wtfast by adding the module as defualt.
thanks.

zw963, 2017-08-16:
> it works well with shadowsocks-libev (ss-redir, transparent proxy)

Which router you use? thanks

> I noticed that the module was already present for router models with WTFast (AC88/AC3100/AC5300). I've revised the original commit to just extend it to other ARM models rather than modify the default options in config_base.6a.

Cool, I think if I buy a AC-88U now, and burn in the latest asuswrt-merlin, i could use tproxy feature now.

zw963, 2017-08-18:
Hi, I plan to buy a AC-88U today, @odkrys , tproxy feature is supported for this device?

odkrys, 2017-08-18:
all arm merlin devices support tproxy by now.
ac56 ac68 ac1900p ac87 ac3100 ac3200 ac88 ac5300.

RMerl, 2017-08-18:
> Cool, I think if I buy a AC-88U now, and burn in the latest asuswrt-merlin, i could use tproxy feature now.

Not right away, 380.67 does not support it.  380.68 will, which should hopefully be released this weekend, unless any last-minute bug appear.

closed 2017-08-18 by RMerl

zw963, 2017-08-18:
> Not right away, 380.67 does not support it. 380.68 will, which should hopefully be released this weekend, unless any last-minute bug appear.

Cool, thanks very much!

zw963, 2017-08-18:
> it works well with shadowsocks-libev (ss-redir, transparent proxy)

@odkrys , i have a question, you use `ss-redir` with `-u` enabled, right?
current, it is not support for AC66U. but i can use `ss-tunnel` with `-u' enabled.

if `ss-redir -u` is support, if means we does not need to start `ss-tunnel -u`, right?

odkrys, 2017-08-18:
yes. ss-redir with udp mode can do relay all of tcp  and udp packet.