Asuswrt-Merlin/asuswrt-merlin.legacy #1450: Update curl to 7.55.0

CorrosiveCat, 2017-08-13:
Seems to add a few features and bug fixes and might be helpful

RMerl, 2017-08-16:
I tried upgrading to 7.55.1, and it causes a kernel panic at boot time.  Could be some internal changes that aren't compatible with rc (which is linked against it).  If the issue lies in the closed source components of rc, then it means it won't be possible to upgrade to any future curl release.  At this time I have no idea as to the cause - debugging the init process is quite difficult since any crash in it will cause an early reboot.

RMerl, 2017-08-16:
libcurl is linked against rc/dsl_fb.o, which is closed source.  That means only Asus would be able to fix this issue.

commit 8ff38d62 (2017-08-16): curl: reverting to 7.54.1, as the new version causes rc to crash at boot time, causing a kernel panic.

rc/dsl_fb.o is linked against libcurl, and this is another closed source piece over which I have no
control...

Revert "curl: Updated to 7.55.1 (closes #1450)"

This reverts commit b8210c6e8e2aa442e285b3ceeb0d55a87761d8ac.
commit b8210c6e (2017-08-16): curl: Updated to 7.55.1 (closes #1450)
closed 2017-08-16 by RMerl