Asuswrt-Merlin/asuswrt-merlin.legacy #1451: Firmware: 380.67 / Router: RT-AC87U - OpenVPN Client - Traffic is NOT blocked when tunnel is down.

singhandsingh, 2017-08-14:
Hi, 

I was trying to solve a DNS related issue and i discovered a potential related to the following setting..

----
Block routed clients if tunnel goes down: Yes
----

I decided to test the above setting by switching off my fiber internet termination box to simulate WAN down. My asus router is still switched on.. and i am continuously pinging... 8.8.8.8. The results are as follows...

#### WHEN STILL CONNECTED TO THE TUNNEL, THE RTT IS HIGH. EXPECTED RESULT ####

Reply from 8.8.8.8: bytes=32 time=471ms TTL=58
Reply from 8.8.8.8: bytes=32 time=598ms TTL=58
Reply from 8.8.8.8: bytes=32 time=341ms TTL=58
......

#### WHEN FIBER TERMINATION BOX IS SWITCHED OFF. EXPECTED RESULT ####

Request timed out.
Request timed out.
Request timed out.
......

#### WHEN FIBER TERMINATION BOX COMES ONLINE. FAIL !!! RTT IS LOW AND THAT MEANS TRAFFIC LEAKS OUTSIDE TUNNEL ####

Reply from 8.8.8.8: bytes=32 time=7ms TTL=48
Reply from 8.8.8.8: bytes=32 time=6ms TTL=48
Reply from 8.8.8.8: bytes=32 time=4ms TTL=48
....

#### SUBSEQUENTLY. AFTER A FEW SECONDS... TRAFFIC NEVER FLOWS THROUGH THE TUNNEL UNTIL I MANUALLY START/STOP THE SERVICE ####

Reply from 192.168.1.1: Destination net unreachable.
Reply from 192.168.1.1: Destination net unreachable.
Request timed out.
Reply from 192.168.1.1: Destination net unreachable.
Request timed out.
Request timed out.
Request timed out.
...
 #####

Two issues here. First, there is a traffic leak when the WAN comes up. 

Secondly, when the WAN comes up, traffic doesn't flow through the tunnel again.

Syst3mSh0ck, 2017-09-27:
Does the VPN tunnel re-establish once the WAN interface comes back up?

singhandsingh, 2017-09-28:
VPN tunnel is reestablished but the routes seems messed up