Asuswrt-Merlin/asuswrt-merlin.legacy #1456: FW 380.67 corrupted RT-AC5300, 5-1 wireless network does not work

meamecum, 2017-08-20:
I've just got my RT-AC5300 from the store, on last thirsday. Upgraded fw to the latest version from Asus site, 380_7743, it worked fine. Then I uploaded Merlin's FW 380.67. After that router began to work unstable: restarted every 1-2 hour. I also noticed that 5-1 network is not active, control channel's list was empty (nothing in wl1_chlist).  I tried to upload Asus/Merlin's FWs in FW restoration mode, it didn't bring 5-1 network back to life. Flashing of 380.68 turned router into a brick. It enters into FW restoration mode if I press on reset during power on, but nothing happens after FW upload completion, I just see the only power LED is on. Looks like CFE should be restored from working rt-ac5300: options for wireless look different for 2.4/5-1/5-2 networks.

meamecum, 2017-08-20:
Update: 360.68 uploaded successfully after restoration to original 380_7743. 5-1 network does not work. wl1_chlist is empty again. System log-wireless log page has minimum data, log itself is not available.

meamecum, 2017-08-20:
wl1_hwaddr (mac address for 5-1) is also empty

Zzyzx-Wolfe, 2017-08-22:
Do you have a USB3 drive connected? USB3 interferes with 5G. Have heard other users that have had 5G-1 disappear whenever a USB3 drive was attached.

meamecum, 2017-08-22:
No, I use ethernet connection to docsis modem only, provided by carrier, working in bridge mode.

meamecum, 2017-08-22:
I've done a reset via WPS button. Router is back to life, but 5-1 network is not available, configuration is missing. And if I make some changes in nvram, like the attempt to set the mac address for wl1 interface, httpd constanly restarts after nvram commit/reboot, main page of web gui looks incomplete.

metekamil, 2017-08-24:
https://github.com/RMerl/asuswrt-merlin/issues/1460

meamecum, 2017-08-24:
I've tried ti restore to 3341, 7378, 7627, 7743 - result is always the same:
Settings like radioids, hwaddr, chlist, ifname, txchain, rxchain, hw_rxchain, hw_txchain are missing or empty for wl1_

metekamil, 2017-08-24:
Yeah I closed that issue, sorry. The problem returned.  Although I still think its DHCP related. Just to be safe I called ASUS and they are sending me a replacement. I'll find out for sure if this is a hardware issue vs software

metekamil, 2017-08-24:
Is your router behind a modem in bridge mode?

meamecum, 2017-08-24:
Yes, correct.
But I also tried to connect wan port of rt-ac5300 to lan port of my old rt-n66u.
rt-ac5300 detected ip addresses conflict, restarted with a new address, 192.168.2.1
and after that result was the same - 5-1 wifi interface didn't work.

meamecum, 2017-08-24:
btw, if you check asus ac routers forum on snbforums, several owners of different models complain to degraded network speed via 5GHz after upgrade to the latest RMerlin's version.

RMerl, 2017-08-25:
This is best discussed on the support forums rather than on this bug tracker as it's unlikely to be a firmware bug (or if it is, it's not something I have any control over as wifi is closed source).  But try this procedure:

1) Turn off the router
2) Unplug the power
3) Turn it back on to drain any residual electrical charge, wait 3-5 seconds
4) Turn it off, and plug the power back
5) Turn it on, and let it boot normally.

If this doesn't bring back your wifi, then there's a good chance one of your radio has died, and you will need to RMA your router.

closed 2017-08-25 by RMerl