Asuswrt-Merlin/asuswrt-merlin.legacy #1462: Wifi Stops Working - RT-AC88U FW 380.68

GodKratos, 2017-08-26:
I have an RT-AC88U that I've just recently installed the Merlin firmware on (previously on stock ASUS WRT).
Everything works fine and fixes the bug where b/g protection couldn't be disabled however the wifi just stops working randomly and I have to reboot the router to re-enable it.

I tried resetting back to factory defaults this morning (since I hadn;t after switching to Merlin) however this afternoon when we got home the Wifi had stopped working again.

Didn't have this problem with stock firmware.

GodKratos, 2017-08-26:
Is it possible to restart the wifi interfaces without rebooting the router?
You can disable and then enable the "Enable Radio" option on the Wireless Professional setting page but that reboots the router completly.

Syst3mSh0ck, 2017-08-26:
Try a Factory Reset first of all.

adamsiem, 2017-08-26:
@GodKratos I have a similar issue with unstable Wifi on an RT-AC88U STOCK Access Point providing signal to an RT-AC66U 380.68 Repeater.

I had a very solid connection for 6 months between an RP-N12 STOCK Access Point > RT-AC66U STOCK Repeater.

I replaced the RP-N12 with an RT-AC88U and now the repeater connection of the RT-AC66U fails several times per day. (NOTE: Repeater Mode only functions over 5Ghz. The 2.4Ghz connection doesn't provide an IP to the client connecting to the Repeater. Both signals have 4 bars in the Repeater Config, and the RSSI is ~-50 with a 450-750mbps connection over 5Ghz WHEN operational.)

One next step is to upgrade the RT-AC88U to Merlin, BUT I'm trying to isolate the unstable wifi issue on the RT-AC66U first. The RT-AC88U is the new addition causing issues.

adamsiem, 2017-08-26:
I did upgrade the RT-AC88U to 360.68 and the issue reported above with the unexpected Wireless Log result remains. Maybe this operates as designed - but I would expect the log to report the broadcast SSID - and note the repeated SSID, versus how it currently reports the repeated SSID and notes the broadcast.

[image]

[image]

[image]

GodKratos, 2017-08-26:
@Syst3mSh0ck in my first post I already mentioned I've done a restore to factory settings already.

This issue is happening quite regularly currently. Going by when the DHCPACK events stop occurring for wireless devices in the system log it appears to be about 6 hours after a reboot before the wireless stops working.

GodKratos, 2017-08-26:
Does anyone know if there is a particular FW version I could try to downgrade to to avoid this issue or do I just need to go back to stock FW for now?

adamsiem, 2017-08-27:
@GodKratos I've been running OK since upgrading both Access Point and Repeater to Merlin 380.68.

GodKratos, 2017-08-27:
Thanks but I'm already on Merlin 380.68 so don't think that's gonna help...

amplatfus, 2017-08-27:
I am on RT-AC88U HW Ver: A2, FW 380.68 for about 1 week and I have no issue.

GodKratos, 2017-08-29:
I've disabled the three options in the AiProtection section and the problem appears to have gone away so it seems to be one of them causing the problem. (Malicious SItes Blocking, Vulnerability Protection & Infected Device Prevention & Blocking).

I had enabled email alerts for them and hadn't received any so I don't think a policy is being triggered, rather there is a bug in one of them.

I'll turn each one on at a time to see which causes the problem.

GodKratos, 2017-09-01:
It appears to be the Vulnerability Protection option causing problems.
I enabled that one on its own and I had problems connecting to wifi. As soon as I turned it off everything started connecting again.

GodKratos, 2017-09-13:
Nope the problem is still occurring with all of the AIProtection options disabled but it seems to be a lot less frequent. I have discovered if I simply change the control channel for the 5GHz network then everything starts connecting again.

fleccy, 2017-09-17:
I've had an issue similar to this but purely on the 2.4ghz wireless networks which only started happening since I upgraded my firmware to 380.68 and set up a VPN. Resetting to factory settings could not resolve this. The issue I was getting was: dns_probe_finished_bad_config

Anyway, after pulling my hair out setting this seems to make it work:

[image]

Not ideal but I do like this feature as I can stop other people in my house looking at things they shouldn't by setting the DNS on their individual devices.

5ghz always worked fine so I don't quite understand why this has started happening. Some of my home devices just stopped working such as Amazon Alexa, My OnePlus X, Canary, which all operate on 2.4ghz only. Some devices however were still working fine on 2.4ghz when I swapped over and I just couldn't understand why. These were compatible with 5ghz also though.

I hope this helps someone!

Thanks

dapidc, 2017-10-20:
I also have same problem with GodKratos here with AC88U. At first only 2.4GHz, so I always could restart router by conneting to 5GHz, but now both have problem. The wifi just stop, like something is telling them to stop. Restarting the router is the only option to get them back to work.
Is it because it is used in office with 30 devices connected? 

Changing router configuration doesn't seem to help.

My firmware version is now 3.0.0.4.382_15850. 

I am suspecting someone is using Draft N or Pre N devices that jammed my router, but 5GHz should still be O.K.

I am out of clue right now. Today I have restart three times in 10 minutes, but after the third we have smooth connections for three hours now...

GodKratos, 2017-11-02:
I've been continuing to have Wifi stopping from working so have gone back to the default stock firmware now.
So far it seems to be stable without the same issues.

physbyte, 2017-11-15:
I also have the same problem, wifi stops working after ~ 1.5-2 day, restarting wifi module solves the problem.

I believe that you need to add an option that at a certain time would reboot the wifi module, think this can be done with the help of a script, but if it were possible in the firmware by default, it would not be superfluous.

**Divice:** Asus N66U
**Firmware:** 380.68_4
**Wifi:** 2.4 GHz
**Channel:** 13

Waghead, 2017-11-18:
I also have this problem.
First i hade stock firmware (ASUS RT-AC88U Firmware version 3.0.0.4.382.15850). Suddenly the 5GHz was disabled/broken, see half wifi-symbol below. I changes to Asuswrt-Merlin (RT-AC88U_382.1_beta4) and in one hour i had the same problem. I now use Asuswrt-Merlin RT-AC88U_382.1_0, still not working for extended time.
[image]

My 2.4GHz wifi works ok. Menus inside router is really slow.

jbombs16, 2017-11-22:
I am having a similar issue where I get the half wifi icon. However my 2.4 stops working and says it is now on channel 0. Then the router soon after locks up and only 2 lights on the router stay solid until I reboot. On lasted non beta with ac3100

dapidc, 2017-11-22:
Hi GodKratos,

You mention about going back to stock firmware. How do we do that? 

[ why am I asking this? Because the problem is now getting more often. I can do restart three times in only 10 minutes Windows. Once 2.4GHz is gone, other time 5GHz is gone. Sometimes both gone, and I have to turn it off to restart ]

Thanks,

Dikirim dari Email untuk Windows 10

Dari: GodKratos
Dikirim: Thursday, November 2, 2017 11:16
Kepada: RMerl/asuswrt-merlin
Cc: dapidc; Comment
Subjek: Re: [RMerl/asuswrt-merlin] Wifi Stops Working - RT-AC88U FW 380.68(#1462)

I've been continuing to have Wifi stopping from working so have gone back to the default stock firmware now.
So far it seems to be stable without the same issues.

Waghead, 2017-11-22:
I returned my broken router today. It's now my 3rd unit. Let's hope this one will make it. 😝

dapidc, 2017-11-24:
No problem with the unit. It’s the firmware.

Now I got current control channel 0, and no way I can change the control channel without restart.

Bad firmware, and nobody care to fix the problem. I shouldn’t have chose ASUS, they keep releasing bad products.

Dari: Waghead
Dikirim: Wednesday, November 22, 2017 15:24
Kepada: RMerl/asuswrt-merlin
Cc: dapidc; Comment
Subjek: Re: [RMerl/asuswrt-merlin] Wifi Stops Working - RT-AC88U FW 380.68(#1462)

I returned my broken router today. It's now my 3rd unit. Let's hope this one will make it. 😝

Waghead, 2017-11-24:
I beg to differ. Clearly it is the hardware when u can't get it to work even though I tried every single software update available. Including  Merlin. At the shop they said it could be bad soldering on the wifi antenna. Maybe it is maybe not. Anyways it not going to fix your broken wifi with software.

Crazar, 2017-12-02:
Got a great deal on this broken RT-3100.   I am getting the half lit icon. 5ghz wont turn on. Tried pressing the button in the front but that only turns on/off the 2ghz wireless.  Have tried 30/30/30 reset, push button reset.  Highly disappointed.   Maybe broken button in the front? DOubt it. There is a setting or something or maybe they produce crap.

RMerl, 2017-12-02:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2017-12-02 by RMerl

GodKratos, 2017-12-03:
How about developing the bugs out of your firmware then?
Stock FW works fine but the Merlin FW causes wifi to stop working constantly.
Seems like an issue that needs to be tracked to me...

dapidc, 2017-12-03:
See?

Nobody cares. Stop buying ASUS product will help you get on with your live
easier.

Their flimsy hardware plus substandard firmware...

jrhortian, 2017-12-04:
I just noticed this issue last night when I went to watch my TV in the living room. I only use 5 Ghz if my devices support it. I was on the latest version of Merlin. After several reboots, I would get the 5 Ghz to work for about 10 minutes, after that the wireless was down. The 2.4 Ghz seemed to work without an issue. After screwing around with it for about 30-60 minutes I reverted back to the current stock firmware and all has been well. Fix this please......

dapidc, 2017-12-12:
Sometimes I got this:

Note: the router you are using is not an ASUS device or has not been authorised by ASUS. ASUSWRT might not work properly on this device.

And there is no way I can login or connect to the router before restarting it.

The SSID are there, sometimes 2.4GHz cannot connect (without error), sometimes 5GHz cannot, sometimes both.

From: jrhortian
Sent: Monday, December 4, 2017 07:31
To: RMerl/asuswrt-merlin
Cc: dapidc; Comment
Subject: Re: [RMerl/asuswrt-merlin] Wifi Stops Working - RT-AC88U FW 380.68(#1462)

I just noticed this issue last night when I went to watch my TV in the living room. I only use 5 Ghz if my devices support it. I was on the latest version of Merlin. After several reboots, I would get the 5 Ghz to work for about 10 minutes, after that the wireless was down. The 2.4 Ghz seemed to work without an issue. After screwing around with it for about 30-60 minutes I reverted back to the current stock firmware and all has been well. Fix this please......

panther7, 2018-01-17:
I have same issue on RT-AC66U (v380.69), but this issue is on original frimware ... :-/

dapidc, 2018-01-18:
Submit it to Service Center. They will replace your unit. If they don’t, Ask them to replace the unit. If they refuse, throw it to them.

Waghead, 2018-01-19:
I agree (to dapidc) , it's broken.