Asuswrt-Merlin/asuswrt-merlin.legacy #1464: Accept certificates with txt extension

FinnishFlash15, 2017-08-31:
In version 380.68 Web UI doesn't accept SSL certificates or private keys that are uploaded with a txt extension. That's how they come from ZeroSSL. We discussed this on IRC and you saw no problem in making a change to accept these.

Consider this a reminder for the next development cycle. :)

unsynaps, 2017-09-05:
Just rename the extension on the cert?

RMerl, 2017-09-05:
It's not obvious that the issue is the extension, router simply rejects the file.  OP knew it was a filename issue because we discussed it together on IRC.

I agree that this is an artificial limitation that probably has no reason to be, and I want to look at possibly removing it in the future.

RMerl, 2017-09-09:
After a quick glance, this might be difficult to implement, as httpd relies on the filename to determine if it`s dealing with a key, a crt, a crl, etc...

FinnishFlash15, 2017-09-11:
If it's difficult to implement, then maybe add a notification to let the user know what the extension needs to be? That'd be enough in my opinion.