Asuswrt-Merlin/asuswrt-merlin.legacy #1466: TrafficAnalyzer Statistic will bomb when it running long time

tldzyx, 2017-09-02:
I noticed that the traffic analyzer statistic feature will create a file in '/jffs/traffic.db', and as time goes on it's size will grown biger than biger, until the `/jffs` partition has no space to store it, then this feature will bomb, and the solution is simply, just delete the '/jffs/traffic.db' then restart traffic analyzer statistic.

[image]

so, I suggest add a button on the `Advanced_System_Content.asp' make it simply.

by the way, if this feature can limit time range for example `last month` or `last week` or `last day`echo , it's feel better.

Syst3mSh0ck, 2017-09-27:
How long does it take for the database to fill up roughly? I suppose this would vary massively depending on the usage type and frequency however I propose an automatic flush when the free space in the JFFS partition reaches a certain threshold such as ~90%.

tldzyx, 2017-10-03:
In my environment, it's will fill up between one to two month, and before I found this reason, I have to reset the router to make it enable.

RMerl, 2017-10-03:
I know Asus added some database cleanup last year, however that code is entirely closed source, so nothing I can do to check if/how it's working.

I'd suggest contacting Asus about this if you see your database grow abnormally large.

turribeach, 2017-12-20:
Hi, is it possible to change the location of traffic.db? If so, where do I do that? I am thinking on mounting an NFS share from my NAS and putting the file there.

unsynaps, 2017-12-20:
@tldzyx
Just make a cron job for a scheduled restart

@turribeach
Could make a symlink. I have done that before to relocate files that dont have settings in their config files.