Asuswrt-Merlin/asuswrt-merlin.legacy #1467: Host names in per device Traffic Manager only for connected devices (380.68)

tprochazka, 2017-09-02:
Ped device Traffic Manager is the one of the best feature in Merlin.
It is great that it display the host name instead of just IP aderess.
But it's weird that it works just if the device is currently connected.
When the devices is disconnected now it display just the IP address.
But the router itself must have persisted MAC -> Host name table, because all devices get the same name after reconnect and also after router restart.
Would be great if this will be used also in the Traffic Manager to allows display all host names correctly.

Btw. Currently I store traffic manager just in the RAM because I don't know what to use in the Custom location, there should be some smart default  ;-).

tprochazka, 2017-09-02:
It looks that it is already solved here
https://github.com/RMerl/asuswrt-merlin/pull/1136

RMerl, 2017-09-09:
Hostnames are not reported on disconnected devices because there's no way to ensure that whatever is using that IP is the exact same device.  Rather than display possibly incorrect data, I stick to what is known to be accurate: the IP itself.

This is by design.

closed 2017-09-09 by RMerl

tprochazka, 2017-09-10:
I don't agree. Also now it can display incorrect data when I download 10 GB by device A and then connect device B with the same IP address. It will now display that device B consumed 10GB. This is exactly the same like you described the scenario. Best would be store statistical data by MAC address and completely ignore the IP address.

tprochazka, 2017-09-10:
But as I see the router use always the same IP address in DHCP mode when for a device where I changed the name manually. So it is not problem to use static MAC -> host name table.