Asuswrt-Merlin/asuswrt-merlin.legacy #1469: Add Documentation That Clearly Explains How to Add Certs

NuttBoxer, 2017-09-05:
I flashed the latest release of Merlin for the n66u this weekend, mainly for the purpose of using the split-tunnel feature for my VPN.  The process was super simple except for one thing.  It took me over an hour to figure out I had to click on the very small yellow text in order to add the signature for the VPN cert.  Every tutorial I saw made reference to a "tab" or a "page".  But it's really a link that opens a pop-up.  To help other users avoid this in the future, I suggest adding documentation around it, or doing something with the link to make it more noticeable.

unsynaps, 2017-09-05:
"very small yellow text"... I have a 4K monitor and its hardly small. Also kind of stands out being yellow.

NuttBoxer, 2017-09-05:
Is that the kind of response I should expect from project devs here?  Arguing with the users?  Maybe if it was the only use of yellow text...  Anyway, this is real feedback from a first time user.  Use it as an opportunity to improve, or don't.

RMerl, 2017-09-05:
@unsynaps is not a developer, he's a regular user like you.

NuttBoxer, 2017-09-05:
Updated comment.  Love the project, was exactly what I needed.  Just don't want other users to go through the same pain I did in this one area.

RMerl, 2017-09-09:
Feel free to update the Wiki if you can improve on the available documentation.  This is a one-man project, I simply don't have the time to handle all the documentation in addition to development, project management, etc...  The majority of the content of the Wiki was provided by the community.

closed 2017-09-09 by RMerl