Asuswrt-Merlin/asuswrt-merlin.legacy #1470: How to set sysctl on asuswrt-merlin?

pmalek, 2017-09-05:
Inspired by this https://cloudplatform.googleblog.com/2017/07/TCP-BBR-congestion-control-comes-to-GCP-your-Internet-just-got-faster.html I decided that I'd like to set up BBR on my Asus RT-AC68U.

To my surprise there is no `/etc/sysctl.conf` nor `/rom/etc/sysctl.conf`.

I've tried looking through this repo but couldn't find anything of substance.

Can you share how to set syctl config on asuswrt-merlin?

amplatfus, 2017-09-06:
Looks interesting. Could you please describe how it can be set up supposing we found sysctl.conf on asus-merlin?
Thanks for idea!

amplatfus, 2017-09-06:
Maybe this should help us: https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/busybox/procps/sysctl.c

pmalek, 2017-09-06:
There's an [article on cyberciti](https://www.cyberciti.biz/cloud-computing/increase-your-linux-server-internet-speed-with-tcp-bbr-congestion-control/) about this.

Basically you have to check if your kernel is compiled with 
```
CONFIG_TCP_CONG_BBR
CONFIG_NET_SCH_FQ
```

(I forgot about this one - maybe asuswrt-merlin doesn't come with those (I cannot check it ATM))
and then you can either add a custom `sysctl.con` in e.g. `/etc/sysctl.d/10-custom-kernel-bbr.conf` or append it to you `/etc/sysctl.conf`

```
sudo sh -c "echo net.core.default_qdisc=fq >> /etc/sysctl.d/10-custom-kernel-bbr.conf"
sudo sh -c "echo net.ipv4.tcp_congestion_control=bbr >> /etc/sysctl.d/10-custom-kernel-bbr.conf"
```

amplatfus, 2017-09-06:
@RMerl  Hi, please it is possible do be done in router directly in terminal?
Many thanks for 360* support and hard work.

closed 2017-09-09 by RMerl