Asuswrt-Merlin/asuswrt-merlin.legacy #1472: Mounting USB HDD not working 380.68 on RT-N66U

fox244243904, 2017-09-07:
- Already restore default settings
- Manually mount gives a "invalid argument" error

Device:
WD Passport 3 TB formatted with NTFS

Atavic, 2017-09-08:
>The Asus router USB 3 ports do not function with all USB 3 hubs and storage devices.

[Source](https://www.snbforums.com/threads/rt-ac68u-wd-my-passport-2tb-usb-3-0-problem.18548/page-2)

Try to enable **Reducing USB 3.0 interference** so the USB 2.0 mode is enforced.

fox244243904, 2017-09-09:
thanks for help.
I was on 380.64 and the HDD working fine.
fdisk -l detects the HDD, i do not use a hub.
nvram variable name "usb_usb3" is not defined. (to disable the USB 3.0, but i do not want to disable it...)

closed 2017-09-09 by RMerl