Asuswrt-Merlin/asuswrt-merlin.legacy #1476: Possible bug in log file display, affecting only some browsers

ShakeThat, 2017-09-13:
When I first load the "System Log" page, the log data is displayed properly. But when the refresh occurs, everything after the first appearance of a "<" in the log data disappears. This behavior began in some recent firmware release, probably 380.68. I suspect it might result from the following line in Main_LogStatus_Content.asp:

    document.getElementById("textarea").innerHTML = logString;

Presumably there is not expected to be markup in the logString, so perhaps this should be setting .value rather than .innerHTML? (I don't claim to know what I'm talking about here; I have no web development experience. I tried playing with this in the TryIt Editor mentioned below, and in my browser, setting .value didn't have the desired effect.)

There is a line in the log that looks like:

    ipt_account 0.1.21 : Piotr Gasidlo <quaker@barbara.eu.org>, http://code.google.com/p/ipt-account/

and I suspect that "<quaker@barbara.eu.org>" is looking like invalid HTML, causing it and everything that follows to disappear. Note that the browser I'm using is quite old, and the problem does not occur with others I've tried, so I don't consider this critical. When I click the "Save" button, the entire log is downloaded, without truncation.

I can replicate this behavior using the TryIt Editor at

    https://www.w3schools.com/jsref/tryit.asp?filename=tryjsref_element_innerhtml

ShakeThat, 2017-09-13:
Of course I meant "&lt;quaker@barbara.eu.org&gt;", ha ha.

Syst3mSh0ck, 2017-09-27:
Thanks for repoting. Can you please advise which browser this impacts?

ShakeThat, 2017-09-27:
The problem occurs with Safari 4.1.3 and OmniWeb 5.11.2, both ancient, running on an ancient Mac (my daily driver, for reasons). Naturally, there had not been a problem until auto-refresh of the log was introduced., so it looked like a bug.

I've since tested it with current versions of Safari, FireFox, Chrome and OmniWeb, all of which do NOT exhibit the problem. Therefore, it's more likely a bug in the old browsers, or maybe a standards change, so I'm going to close this. Thanks for looking!

closed 2017-09-27 by ShakeThat