Asuswrt-Merlin/asuswrt-merlin.legacy #1479: snmpd reporting wrong interface speed

freedomkk-qfeng, 2017-09-18:
I'm using snmp to report devices status. Regarding the interfaces speed (ifSpeed parameter) ASUSWRT is the only that is reporting all the interfaces as 10 Mbps (ifSpeed = 10000000,ifHighSpeed = 10) when they should report 1Gbps. Like this
```
# snmpwalk -v 2c -c public 192.168.2.2 1.3.6.1.2.1.31.1.1.1.15
iso.3.6.1.2.1.31.1.1.1.15.1 = Gauge32: 10
iso.3.6.1.2.1.31.1.1.1.15.2 = Gauge32: 10
iso.3.6.1.2.1.31.1.1.1.15.3 = Gauge32: 0
iso.3.6.1.2.1.31.1.1.1.15.4 = Gauge32: 0
iso.3.6.1.2.1.31.1.1.1.15.5 = Gauge32: 10
iso.3.6.1.2.1.31.1.1.1.15.6 = Gauge32: 10
iso.3.6.1.2.1.31.1.1.1.15.7 = Gauge32: 0
# snmpwalk -v 2c -c public 192.168.2.2 1.3.6.1.2.1.2.2.1.5
iso.3.6.1.2.1.2.2.1.5.1 = Gauge32: 10000000
iso.3.6.1.2.1.2.2.1.5.2 = Gauge32: 10000000
iso.3.6.1.2.1.2.2.1.5.3 = Gauge32: 0
iso.3.6.1.2.1.2.2.1.5.4 = Gauge32: 0
iso.3.6.1.2.1.2.2.1.5.5 = Gauge32: 10000000
iso.3.6.1.2.1.2.2.1.5.6 = Gauge32: 10000000
iso.3.6.1.2.1.2.2.1.5.7 = Gauge32: 0
```
It seems a bug, will it be fix in the future?
My device is RT-AC66U，fireware version is 380.68_2

Syst3mSh0ck, 2017-09-27:
Looks like the OIDs are potentially querying the incorrect variables. Further investigation is required. Ethernet Link Speed can be retrieved from the following location as follows: cat /sys/class/net/<interface>/speed

RMerl, 2017-10-11:
SNMP retrieves the info from the kernel.  It's possible that Broadcom's proprietary driver does not properly report that information, in which case there's not much that can be done about it.

closed 2017-10-11 by RMerl