Asuswrt-Merlin/asuswrt-merlin.legacy #1480: Problems with firmware 380.68 and 380.68_2

pedja1975, 2017-09-18:
I have installed firmware 380.67 and everything works correctly but when updating to firmware 380.68 or 380.68_2 I have several faults. I have installed entware and in the transmission that when starting the router starts transmission but with the new firmware I have to start it manually by terminal. I have an ATA switchboard for the VOIP that with the new firmware does not register the phone. All this works very well with the firmware 380.67.

Syst3mSh0ck, 2017-09-27:
Thanks for reporting. I would recommend downgrading to 380.67 until the 380.68 release is more stable. Alternatively you could try using John's fork which includes the latest upstream security patches but retains the older codebase for stability reasons.

RMerl, 2017-10-07:
Do a factory default reset and reconfigure.  It's unlikely to be a firmware issue since it works fine for everyone else.  I have both an ATA and an IP phone myself (on a separate provider), both are working fine.

closed 2017-10-07 by RMerl