Asuswrt-Merlin/asuswrt-merlin.legacy #1481: Wonderful "No internet, secured" problem

ArchaicStuff, 2017-09-18:
I live in an apartment that pays for our internet, I don't have my own modem and plug directly into the wall.
I installed Merlin to help with pesky ip conflicts caused by I still don't know what. New problem though, every morning I have to resetup my router to get any internet access and have to set my router as an access point for anything to work. Asus customer support is useless so I figured I'd check to see if it was a Merlin issue.

Syst3mSh0ck, 2017-09-27:
You need to disable DHCP because the upstream Edge Router in your apartment will already been handling this functionality. It is never a good idea to have multiple DHCP servers running within the same subnet. Effectively you want the Router to act as a dumb switch with WAP capabilities. Hope this helps. Any questions let me know. Thanks.

RMerl, 2017-10-07:
Best you get this reviewed on the support forums first.

closed 2017-10-07 by RMerl