Asuswrt-Merlin/asuswrt-merlin.legacy #1482: Misleading OpenVPN "Cipher Negotiation" option

julijonas, 2017-09-18:
It would make sense that "Cipher Negotiation" set to "Enable (with fallback)" uses the set fallback and "Enable" does not use any fallback.

However, in reality "Enable" omits the "cipher" config entry and by default `cipher` is set to insecure BF-CBC. I can establish BF-CBC data channel using `--ncp-disable` in client. So while users might use "Enable" when hardening their config in fact they weaken it.

See `--cipher alg` defaults in
https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
Default behaviour explained here
http://openvpn-users.narkive.com/4EI6CfG0/correct-use-of-ncp-ciphers-ncp-disable-for-the-data-channel-cipher

It would also be great to expand "Legacy/fallback cipher" to include all ciphers, especially AES-*-GCM ones, which as I understand are current best practice.

Syst3mSh0ck, 2017-09-27:
Thanks for reporting this. Hopefully it will be fixed when Merlin has the time to look into it.

RMerl, 2017-10-07:
Most of these is handled by OpenVPN itself and not the firmware.  The NCP implementation was done based on the OpenVPN 2.4 manual documentation.  If default OpenVPN behaviour is undesirable, then this would have to be addressed upsteram.

closed 2017-10-07 by RMerl