Asuswrt-Merlin/asuswrt-merlin.legacy #1485: DDNS error: no-ip.com

VNGXR, 2017-09-21:
> Sep 21 23:54:27 ddns_update: ez-ipupdate: starting...
> Sep 21 23:54:28 ddns_update: connected to dynupdate.no-ip.com (8.23.224.120) on port 80.
> Sep 21 23:54:28 ddns_update: request successful
> Sep 21 23:54:28 ddns_update: asusddns_update: 0

However it shows "ddns_fail_hint" on index page with "Request error! Please try again".

Syst3mSh0ck, 2017-09-27:
Can you include a screenshot of the issue reported please? Thanks.

VNGXR, 2017-09-27:
On the index page: [image]
On WAN - DDNS page: 
[image]

RMerl, 2017-10-11:
Can't reproduce here (also using No-IP).

closed 2017-10-11 by VNGXR