Asuswrt-Merlin/asuswrt-merlin.legacy #1487: EA-AC87

ersingencturk, 2017-09-24:
can we use this with EA-AC87 ?

RMerl, 2017-09-25:
No.

https://github.com/RMerl/asuswrt-merlin/wiki/Supported-Devices

closed 2017-09-25 by RMerl

ersingencturk, 2017-09-25:
it's the list of supported devices, and I already checked it before posting here. it might be unsupported, but it's technically similar hardware architecture with  RT models and runs asus firmware as far as I know.what's the reason why this can not be run on this model ? I might help for code changes if you can shed some light instead of just answering "no"

RMerl, 2017-09-25:
That device is not supported for many reasons, but the most important ones:

1) The majority of my enhancements target specifically router mode, so they won't be available in bridge/ap/repeater mode.  That means there is little point in having Asuswrt-Merlin support it.

2) The last firmware release (with a GPL drop) from Asus for that model is nearly two years old.  For each supported model, Asuswrt-Merlin requires closed source components from a compatible GPL release (that means usually released by Asus within at most a month or two from one another).  The closed source components from that 2015 release cannot be used with Asuswrt-Merlin's GPL 380_7743 code.

So even if another developer wanted to devote the time to support it, it's not possible from a technical point of view, because the binary blobs from that 2 years old 374_xxxx GPL release cannot be used with Asuswrt-Merlin's code that's based on 380_7443.

ersingencturk, 2017-09-25:
I see, this was enlightening, thanks for the reply.