Asuswrt-Merlin/asuswrt-merlin.legacy #1488: Use a proper button for submitting the login form

chrboe, 2017-09-25:
Currently the login form is submitted via a `div` element that's disguised as a button:

    <div class="button" onclick="login();">Sign in</div>

This breaks auto-submitting of the form. Extensions such as [browserpass](https://github.com/dannyvankooten/browserpass) depend on `<input type="submit">` elements being present to automatically fill and submit password forms. It would thus be better to use a proper `submit` button to ensure compatibility with such extensions.

(See also https://github.com/dannyvankooten/browserpass/issues/156)

Syst3mSh0ck, 2017-09-27:
Agreed this would be a nice fix which appears to be quite trivial.

RMerl, 2017-10-07:
Not sure what was Asus's reason behind that design decision - perhaps so it can more easily be disabled if an invalid firmware is detected?

I'd need to get a clearer idea as to the reason behind their decision before changing that.

alindt, 2017-10-07:
After many years of debugging Chinese software, I'm pretty sure it is just a weird way of achieving a simple task, with no real design decision behind

RMerl, 2017-10-10:
Fixed on AM382, will consider backporting it to this branch based on future release plans for a 380.xx release.

commit f785f26d (2017-11-25): webui: make 'sign in' a button instead of <div> (fixes #1488)
closed 2017-11-25 by RMerl