Asuswrt-Merlin/asuswrt-merlin.legacy #1493: 5GHz band is gone on 380.68_2 - Asus RT-AC66U

szalap, 2017-09-30:
Hi,

After upgrading to the latest and greatest I have lost my 5GHz network. I have checked it on multiple devices and the 5GHz was nowhere to be found. 
After downgrading to 380.68_0 everything is back to normal and I have both 2.4GHz and 5GHz available.

alindt, 2017-09-30:
@szalap
I'm using 380.68_2 on a RT-AC66U_B1 and the 5GHz band works fine.
Do you have the A1 or B1 hw revision?

szalap, 2017-09-30:
It's A1 revision, so I guess here might be the issue.

gschukin, 2017-10-07:
Have the same problem with RT-N66U. 5GHz doesn't gone completely, but available from time to time

RMerl, 2017-10-07:
Not a firmware issue, especially as there's been zero wifi-related changes between 380.68_0 and 380.68_2.

closed 2017-10-07 by RMerl