Asuswrt-Merlin/asuswrt-merlin.legacy #1502: MiniDLNA 1.2.1 released

fmpallini, 2017-10-06:
Some nice bugfixes, if easy to merge.

1.2.1 - Released 24-Aug-2017
--------------------------------
- Added Movian client detection and subtitle support.
- Fixed an issue with discovery on non-Linux systems.
- Fixed Bonjour discovery compatibility with TiVo Bolt.
- Fixed NFO file parsing, and added change monitoring support for them.
- Added a workaround for video thumbnails on some Samsung clients.
- Added DoS protection for event subscriptions.
- Fixed content browsing issues with some Samsung TVs.
- Improved non-destructive update scan support.

RMerl, 2017-10-07:
Note that Asuswrt currently uses a snapshot that lies somewhere between 1.2.0 and 1.2.1, so some of these changes are already included.

RMerl, 2017-11-22:
There are too many changes in 1.2.1, combined with Asus's being now too different from upstream.  This makes it too complicated to merge the two together, so I gave up on it.  Merging will have to come from upstream.

closed 2017-11-22 by RMerl