Asuswrt-Merlin/asuswrt-merlin.legacy #1503: httpds high cpu consumtion

gschukin, 2017-10-07:
I'm having RT_N66U with 380.64_2. When I access web UI it is unresponsive and httpds process consumes almost all cpu.
[image]

francoisjacques, 2017-10-22:
I'm seeing this as well. Same router model. It's been around 50-80% for the last minutes. I'm actually thinking about disabling it completely until this is resolved. If there's anything we can do to help...

GitzJoey, 2017-11-18:
i'm using RT-N66U with firmware 380.64_4
[image]

gschukin, 2017-11-18:
@GitzJoey did you login into UI?

GitzJoey, 2017-11-18:
with login into UI, there's high jump (34%) about 1 sec but after that stable around 0.3 - 0.9

 284     1 admintkb S     4980  2.0   0  0.9 httpd -i br0

gschukin, 2017-11-18:
Just checked (after login)
Firefox cause spikes up to 70%
Chrome cause constant CPU consumption >90%