Asuswrt-Merlin/asuswrt-merlin.legacy #1505: Adding Manually Assigned IP corrupts web page and host list

DFFRZ, 2017-10-10:
ASUS RT-AC68P running 380.68_4

On DHCP Server tab of LAN settings, adding a new entry to "Manually Assigned IP around the DHCP list" causes existing list to be cleared/emptied and web page to become corrupted. Subsequently, only the tab pane will be presented by Advanced_DHCP_Content.asp and the list is empty. The header, other tabs and left navigation pane are not displayed. Condition persists even after reboot.

To recover, I had to revert to 3.0.0.4.380_7743 and restore settings that were saved at 380_7743, then reinstall 380.68_4 and restore settings saved at 380.68_4

Problem is reproducible.

RMerl, 2017-10-11:
This usually occurs with a corrupted entry, for instance one containing a quote in your description.  the webui is already trying to filter out those characters, but some browsers manage to still bypass those validation filters.

closed 2017-10-11 by RMerl

DFFRZ, 2017-10-11:
That was the problem. I had a single quote mark in one of the Hostnames/descriptions. Thanks.