Asuswrt-Merlin/asuswrt-merlin.legacy #1510: RT-AC87U panics when a torrent download starts

imavroukakis, 2017-10-13:
I've noticed that if I start download a torrent from my Synology NAS, the router will frequently panic and reboot. Once it has rebooted, the torrent is downloaded normally and I don't experience another crash. However, after approximately 24hrs, the same issue will occur when another torrent is started. I am running 380.68_4 and I can't see anything in the logs that points out to the root cause.
All I see in the logs around the time the hard crash happens is this

```
Oct 12 22:28:04 miniupnpd[13567]: Failed to get IP for interface ppp0
Oct 12 22:28:05 openvpn[9588]: event_wait : Interrupted system call (code=4)
Oct 12 22:28:09 openvpn[9617]: write UDP: Network is unreachable (code=101)
Oct 12 22:28:16 openvpn[9617]: event_wait : Interrupted system call (code=4)
Aug  1 03:00:14 kernel: pflash: found no supported devices
Aug  1 03:00:14 kernel: bcmsflash: found no supported devices
Aug  1 03:00:33 Mastiff: init
Aug  1 03:00:41 kernel: xhci_hcd 0000:00:0c.0: Failed to enable MSI-X
Aug  1 03:00:41 kernel: xhci_hcd 0000:00:0c.0: failed to allocate MSI entry
Oct 12 22:30:44 ddns_update: ez-ipupdate: starting...
```

imavroukakis, 2017-10-13:
What application are you talking about?

imavroukakis, 2017-10-13:
Oh I see what you mean @Atavic . No i've not aborted anything, and this has nothing to do with openvpn, this is just what the router logs before crash restarting.

Atavic, 2017-10-13:
OK I looked around and there are a lot of questions regarding it. I can only speculate as I can't find answers around yet. Has your torrent client a good setup regarding ports? I had similar problems in the past and it was caused by a firewall rule - created by me - that I ignored was active. Hope it helps.

unsynaps, 2017-10-13:
Not sure how much help it is but I have a RT-AC87R and a Synology NAS. Have not run into this issue.

All these versions should be current as of this post.

Router: RT-AC87R
Merlin Firmware: 380.68_4
NAS Model: DS1812+
NAS OS Version: DSM 6.1.3-15152 Update 7
DownloadStation Version: 3.8.6-3481

RMerl, 2017-10-13:
Check your router uptime.  This looks more like your Internet connection is dropping, not the router crashing or rebooting.  If that's the case, issue would probably be the modem or the ISP.

imavroukakis, 2017-10-14:
@RMerl first thing I checked. This is just the incidental log around the crash time. I know the router crashes because I've visually verified it restarting when the loss of link occurs, as everything else goes with if , i.e. WiFi disappears etc.

imavroukakis, 2017-10-14:
And also, the uptime is reset as you've mentioned.

imavroukakis, 2017-10-14:
To prove my point, I just replicated the events leading up to the crash
```
Oct 14 18:00:00 rc_service: service 16234:notify_rc start_vpnclient1
Oct 14 18:00:00 rc_service: service 16235:notify_rc start_vpnclient2
Oct 14 18:00:00 rc_service: waitting "start_vpnclient2" via  ...
Oct 14 18:00:00 syslog: VPN_LOG_NOTE: 86: VPN Client 2 already running...
Oct 14 18:00:01 syslog: VPN_LOG_NOTE: 86: VPN Client 1 already running...
Oct 14 18:03:00 crond[484]: USER jander pid 16983 cmd service start_vpnclient1
Oct 14 18:03:01 rc_service: service 16985:notify_rc start_vpnclient1
Oct 14 18:03:01 syslog: VPN_LOG_NOTE: 86: VPN Client 1 already running...
Aug  1 03:00:14 syslogd started: BusyBox v1.25.1
Aug  1 03:00:14 kernel: klogd started: BusyBox v1.25.1 (2017-10-04 15:01:50 EDT)
Aug  1 03:00:14 kernel: Linux version 2.6.36.4brcmarm (merlin@ubuntu-dev) (gcc version 4.5.3 (Buildroot 2012.02) ) #2 SMP PREEMPT Wed Oct 4 15:11:10 EDT 2017
Aug  1 03:00:14 kernel: CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c53c7f
Aug  1 03:00:14 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Aug  1 03:00:14 kernel: Machine: Northstar Prototype
Aug  1 03:00:14 kernel: Ignoring unrecognised tag 0x00000000
Aug  1 03:00:14 kernel: bootconsole [earlycon0] enabled
Aug  1 03:00:14 kernel: Memory policy: ECC disabled, Data cache writealloc
Aug  1 03:00:14 kernel: MPCORE found at 19020000
Aug  1 03:00:14 kernel: On node 0 totalpages: 65536
Aug  1 03:00:14 kernel:   DMA zone: 256 pages used for memmap
Aug  1 03:00:14 kernel:   DMA zone: 0 pages reserved
Aug  1 03:00:14 kernel:   DMA zone: 32512 pages, LIFO batch:7
Aug  1 03:00:14 kernel:   Normal zone: 4864 pages used for memmap
Aug  1 03:00:14 kernel:   Normal zone: 27904 pages, LIFO batch:7
Aug  1 03:00:14 kernel: PERCPU: Embedded 7 pages/cpu @c8215000 s5472 r8192 d15008 u65536
```

imavroukakis, 2017-10-30:
@RMerl is there some way to dump the kernel panic somewhere when this happens? Logs obviously don't catch it.

RMerl, 2017-10-30:
You need a serial cable attached to the pcboard.

ConnorED, 2017-11-12:
I have the same issue with Asus Firmware 3.0.0.4.380_7743
Nov 11 14:21:12 miniupnpd[22048]: upnp_event_process_notify: connect(10.1.1.44:2869): Connection timed out
Nov 11 15:01:54 kernel: htb: htb qdisc 10: is non-work-conserving?
Jul 31 20:00:10 syslogd started: BusyBox v1.17.4
Jul 31 20:00:10 kernel: klogd started: BusyBox v1.17.4 (2017-06-01 23:40:39 CST)
Jul 31 20:00:10 kernel: Linux version 2.6.36.4brcmarm (root@asus) (gcc version 4.5.3 (Buildroot 2012.02) ) #1 SMP PREEMPT Thu Jun 1 23:49:59 CST 2017
Jul 31 20:00:10 kernel: CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c53c7f
Jul 31 20:00:10 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jul 31 20:00:10 kernel: Machine: Northstar Prototype
Jul 31 20:00:10 kernel: Ignoring unrecognised tag 0x00000000
Jul 31 20:00:10 kernel: Memory policy: ECC disabled, Data cache writealloc
Jul 31 20:00:10 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 60416
Jul 31 20:00:10 kernel: Kernel command line: root=/dev/mtdblock2 console=ttyS0,115200 init=/sbin/preinit earlyprintk debug
Jul 31 20:00:10 kernel: Memory: 255512k/255512k available, 6632k reserved, 0K highmem
Jul 31 20:00:10 kernel: Virtual kernel memory layout:
Jul 31 20:00:10 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Jul 31 20:00:10 kernel:     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
Jul 31 20:00:10 kernel:     DMA     : 0xf7e00000 - 0xffe00000   ( 128 MB)
Jul 31 20:00:10 kernel:     vmalloc : 0xd0800000 - 0xf0000000   ( 504 MB)
Jul 31 20:00:10 kernel:     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
Jul 31 20:00:10 kernel:     modules : 0xbf000000 - 0xc0000000   (  16 MB)
Jul 31 20:00:10 kernel:       .init : 0xc0008000 - 0xc003d000   ( 212 kB)
Jul 31 20:00:11 kernel:       .text : 0xc003d000 - 0xc03a5000   (3488 kB)
Jul 31 20:00:11 kernel:       .data : 0xc03bc000 - 0xc03df160   ( 141 kB)
Jul 31 20:00:11 kernel: External imprecise Data abort at addr=0x0, fsr=0x1c06 ignored.
Jul 31 20:00:11 kernel: Mount-cache hash table entries: 512
Jul 31 20:00:11 kernel: CPU1: Booted secondary processor
Jul 31 20:00:11 kernel: Found a AMD NAND flash:
Jul 31 20:00:11 kernel: Total size:  128MB
Jul 31 20:00:11 kernel: Block size:  128KB
Jul 31 20:00:11 kernel: Page Size:   2048B
Jul 31 20:00:11 kernel: OOB Size:    64B
Jul 31 20:00:11 kernel: Sector size: 512B
Jul 31 20:00:11 kernel: Spare size:  16B
Jul 31 20:00:11 kernel: ECC level:   8 (8-bit)
Jul 31 20:00:11 kernel: Device ID: 0x 1 0xf1 0x80 0x1d 0x 1 0xf1
Jul 31 20:00:11 kernel: bio: create slab <bio-0> at 0
Jul 31 20:00:11 kernel: PCI: no core
Jul 31 20:00:11 kernel: PCI: no core
Jul 31 20:00:11 kernel: PCI: Fixing up bus 0
Jul 31 20:00:11 kernel: PCI: Fixing up bus 0
Jul 31 20:00:11 kernel: PCI: Fixing up bus 1
Jul 31 20:00:11 kernel: VFS: Disk quotas dquot_6.5.2
Jul 31 20:00:11 kernel: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
Jul 31 20:00:11 kernel: pflash: found no supported devices
Jul 31 20:00:11 kernel: bcmsflash: found no supported devices
Jul 31 20:00:11 kernel: Boot partition size = 524288(0x80000)
Jul 31 20:00:11 kernel: lookup_nflash_rootfs_offset: offset = 0x200000
Jul 31 20:00:11 kernel: nflash: squash filesystem with lzma found at block 28
Jul 31 20:00:11 kernel: Creating 4 MTD partitions on "nflash":
Jul 31 20:00:11 kernel: 0x000000000000-0x000000080000 : "boot"
Jul 31 20:00:11 kernel: 0x000000080000-0x000000200000 : "nvram"
Jul 31 20:00:11 kernel: 0x000000200000-0x000004000000 : "linux"
Jul 31 20:00:11 kernel: 0x00000039b01c-0x000004000000 : "rootfs"
Jul 31 20:00:12 stop_nat_rules: apply the redirect_rules!
Jul 31 20:00:12 kernel: Registering the dns_resolver key type
Jul 31 20:00:12 kernel: Spare area=64 eccbytes 56, ecc bytes located at:
Jul 31 20:00:12 kernel:  2 3 4 5 6 7 8 9 10 11 12 13 14 15 18 19 20 21 22 23 24 25 26 27 28 29 30 31 34 35 36 37 38 39 40 41 42 43 44 45 46 47 50 51 52 53 54 55 56 57 58 59 60 61 62 63
Jul 31 20:00:12 kernel: Available 7 bytes at (off,len):
Jul 31 20:00:12 kernel: (1,1) (16,2) (32,2) (48,2) (0,0) (0,0) (0,0) (0,0) 
Jul 31 20:00:12 kernel: Options: NO_AUTOINCR,NO_READRDY,BBT_SCAN2NDPAGE,
Jul 31 20:00:12 kernel: Creating 2 MTD partitions on "brcmnand":
Jul 31 20:00:12 kernel: 0x000004000000-0x000007ec0000 : "brcmnand"
Jul 31 20:00:12 kernel: 0x000007ec0000-0x000008000000 : "asus"
Jul 31 20:00:12 qcsapi: write qcsapi conf ok
Jul 31 20:00:12 WAN Connection: Fail to connect with some issues.
Jul 31 20:00:12 kernel: VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
Jul 31 20:00:12 kernel: ctf: module license 'Proprietary' taints kernel.
Jul 31 20:00:12 kernel: Disabling lock debugging due to kernel taint
Jul 31 20:00:12 kernel: et_module_init: passivemode set to 0x0
[... 71 lines cut ...]
Nov 11 19:20:13 rc_service: ntp 873:notify_rc restart_diskmon
Nov 11 19:20:13 disk_monitor: Finish
Nov 11 19:20:13 miniupnpd[886]: version 1.9 started
Nov 11 19:20:14 miniupnpd[886]: HTTP listening on port 44093
Nov 11 19:20:14 miniupnpd[886]: Listening for NAT-PMP/PCP traffic on port 5351
Nov 11 19:20:14 ddns update: Asus update entry:: return: HTTP/1.1 200 OK^M Date: Sat, 11 Nov 2017 23:20:10 GMT^M Server: Apache^M X-Powered-By: PHP/5.6.30^M Content-Length: 0^M Connection: close^M Content-Type: text/html; charset=UTF-8^M ^M
Nov 11 19:20:14 ddns update: retval= 0, ddns_return_code (,200)
Nov 11 19:20:14 ddns update: asusddns_update: 0
Nov 11 19:20:15 ddns: ddns update ok
[... cut ...]
Nov 11 19:20:23 miniupnpd[1261]: HTTP listening on port 50468
Nov 11 19:20:23 miniupnpd[1261]: Listening for NAT-PMP/PCP traffic on port 5351
Nov 11 19:20:28 qtn: bootcfg.tgz exists
Nov 11 19:20:33 crond[464]: time disparity of 1200920 minutes detected
Nov 11 19:20:35 dfs: start dfs scan
Nov 11 19:20:35 nodfs_scan: complete
Nov 11 19:21:58 hour monitor: ntp sync fail, will retry after 120 sec
Nov 11 19:24:07 rc_service: httpds 462:notify_rc restart_sendmail
Nov 11 19:24:11 feedback: wan0=[wan], wan1=[none]

focuszed, 2017-12-08:
Same here, switched back to regular asuswrt and now it's all okay.

imavroukakis, 2017-12-08:
@RMerl good suggestion but I don't fancy my chances not bricking my router. Wasn't there some kernel toggle that allows you to dump panics to an attached FS?

imavroukakis, 2017-12-08:
@focuszed not an option for me unfortunately, thanks though for the tip.

dawgtx, 2018-01-03:
Any solution. I found that on 380.69 the kernel panics as well

focuszed, 2018-01-03:
I recently reverted back to asuswrt and still am getting torrent panics. Guess it's a faulty router.

imavroukakis, 2018-01-04:
@RMerl is it possible to run a debug kernel as described here ? https://kaiwantech.wordpress.com/2017/07/13/setting-up-kdump-and-crash-for-arm-32-an-ongoing-saga/

RMerl, 2018-01-05:
You need to use a serial cable hooked to the router to debug kernel crashes.

ConnorED, 2018-01-06:
Tempted to set up the serial cable. Tired of this issue. Is there a guide to doing it?

imavroukakis, 2018-01-06:
@RMerl yes you've said as much here https://github.com/RMerl/asuswrt-merlin/issues/1510#issuecomment-340443495 . My point is that I don't fancy breaking out the screwdriver every time there's a kernel panic, and in general, there should be an easier way to debug. 
My thoughts revolve around making a debug kernel build available, which will dump the panic log on a properly formatted and attached USB stick for offline inspection

@ConnorED https://wiki.openwrt.org/toh/asus/rt-ac87u#jtag you'll need an appropriate cable.