Asuswrt-Merlin/asuswrt-merlin.legacy #1511: WPA2: Broken with KRACK. What now?

arcao, 2017-10-16:
At the moment no details published yet, they are expected later today.

> So it seems that a flaw in the RNG of WPA2 has been discovered. As I understand, the issue has been reported to vendors in August, but public disclosure will be in a few hours. According to some guys on Twitter, well should pay attention to CVE-2017-13077, 13078, 13079, 13080, 13081, 13082, 13084, 13086, 13087, 13088

Source:
- [WPA2: Broken with KRACK. What now?](https://www.alexhudson.com/2017/10/15/wpa2-broken-krack-now/) by Alex Hudson
- [This is a core protocol-level flaw in WPA2 wi-fi and it looks bad](https://www.reddit.com/r/sysadmin/comments/76lj5q/this_is_a_core_protocollevel_flaw_in_wpa2_wifi/) -
 thread at Reddit sysadmin
- https://www.krackattacks.com/ - website for this WPA2 protocol-level flaw

Kebap, 2017-10-16:
The third website you mention already published details a few hours early now.

justmwa, 2017-10-16:
https://w1.fi/security/2017-1/

deelowe, 2017-10-16:
Much more informed discussions going on over at hacker news: https://news.ycombinator.com/item?id=15481819

Looks like the majority of home routers won't be vulnerable. It's mostly a side client issue.

t-nelson, 2017-10-16:
@deelowe unless they're running in repeater or bridge modes.  Of which ASUSWRT supports both.
```
RT-AC68U-B590:/tmp/home/root# wpa_supplicant 
wpa_supplicant v0.6.10
Copyright (c) 2003-2009, Jouni Malinen <j@w1.fi> and contributors

[snip]
```
Hopefully there are some backports on that :)

deelowe, 2017-10-16:
Does merlin support 802.11r? Looks like that's vulnerable.

Egorvoz, 2017-10-16:
Will someone create the update for routers ? Or what we need to do to protect WiFi ?

t-nelson, 2017-10-16:
@deelowe I didn't see it in any of the settings on my AC68 (AP).  Not by "802.11r", "Fast BSS Transitioning" or "Fast Roaming" anyhow.  There is a "Roaming Assistant" setting in Wireless > Professional, which is the closest thing I found, but is disabled by default.  No wireless settings are exposed in Media Bridge on my AC66... :/

@Egorvoz This is 99% a client issue.  If you are not running a router as a client (repeater or media bridge) then patching your AP is not a high priority.  There is however a slim chance that some AP (mis-) implementations are actively "exploiting" this by accident.  This is not intended to fear monger. Your AP would be functioning normally, just reusing keys.  An attacker would still have to control the AP somehow to do any harm.
TLDR; Worry about patching clients.

Egorvoz, 2017-10-16:
@t-nelson thanks a lot.

Ryujinra, 2017-10-18:
There is at least one patch upstream that has an AP-side workaround for unpatched clients:

https://w1.fi/cgit/hostap/commit/?id=6f234c1e2ee1ede29f2412b7012b3345ed8e52d3

An option to enable or disable this feature in the GUI could be useful, as LEDE has done in these commits: 

https://github.com/lede-project/source/commit/c5f97c9372da3229350184fb263c97d9ea8944c5
https://github.com/lede-project/source/commit/2127425434046ae2b9f02fdbbdd37cac447af19c

RMerl, 2017-10-19:
Asuswrt does not use hostpad.

haarp, 2017-11-02:
Fixed, apparently.

https://www.asus.com/us/Networking/RTAC66U/HelpDesk_BIOS/

RMerl, 2017-11-02:
No GPL available for the N66U/AC66U yet however, so we'll have to wait for it to become available.

The RT-AC86U GPL was released a few days ago, new SDK was merged in 382.1 Beta 3.

zoon01, 2017-11-20:
GPL released N66U
http://dlcdnet.asus.com/pub/ASUS/wireless/RT-N66U_B1/GPL_RT_N66U_30043808120.zip?_ga=2.2389680.1238412958.1511215063-1489479248.1502294606

GPL released AC66U
http://dlcdnet.asus.com/pub/ASUS/wireless/RT-AC66U/GPL_RT_AC66U_30043808120.zip?_ga=2.187923880.1238412958.1511215063-1489479248.1502294606

RMerl, 2017-11-20:
SDK update for AC66U and N66U were already merged with commit https://github.com/RMerl/asuswrt-merlin/commit/c3b42ee79746f38fd6c80a1e5c48a8131fb22c1d .

Note that the RT-N66U GPL is corrupted (file is truncated), but luckily the N66U wireless driver was also present in the RT-AC66U GPL.

Test builds integrating the updated SDK are currently available on https://asuswrt.lostrealm.ca/test-builds .

kernelOfTruth, 2017-11-22:
Lots of devices in their default configuration supposedly aren't affected by KRACK:

https://www.asus.com/Static_WebPage/ASUS-Product-Security-Advisory

```
**10/31/2017 Update on security advisory for the vulnerability of WPA2 protocol ∇**
ASUS is working closely with chipset suppliers to resolve the vulnerability in the WPA2 security protocol, which affects some but not all ASUS products (check the list below). KRACK can exploit the vulnerability only under certain conditions highlighted in the previous update. Your network configuration is more secure when under these conditions:

(1) Routers and gateways working in their default mode (Router Mode) and AP Mode.
(2) Range extenders working in AP Mode. 
(3) When Powerline adapters and switch products are used. 

ASUS is working actively towards a solution, and will continue to post software updates. Find out more: https://www.asus.com/support/ 

Full list of routers unaffected by KRACK while in default mode: 
GT-AC5300
RT-AC1200
RT-AC1200G
RT-AC1200G Plus
RT-AC1200HP
RT-AC1300HP
RT-AC1900
RT-AC1900P
RT-AC3100
RT-AC3200
RT-AC51U
RT-AC52U
RT-AC53
RT-AC5300
RT-AC53U
RT-AC54U
RT-AC55U
RT-AC55UHP
RT-AC56S
RT-AC56U
RT-AC58U
RT-AC66U
RT-AC66U B1
RT-AC66W
RT-AC68P
RT-AC68UF
RT-AC68W
RT-AC86U
RT-AC87U
RT-AC88U
RT-ACRH17
RT-ACRH13
RT-N10P V3
RT-N11P B1
RT-N12 D1
RT-N12 VP B1
RT-N12+
RT-N12+ B1
RT-N12E C1
RT-N12E_B1
RT-N12HP B1
RT-N14U
RT-N14UHP
RT-N16
RT-N18U
RT-N300 B1
RT-N56U
RT-N56U B1
RT-N65U
RT-N66U
RT-N66W
BRT-AC828
DSL-AC87VG
DSL-AC52U
DSL-AC55U
DSL-AC56U
DSL-AC68R
DSL-AC68U
DSL-N10_C1
DSL-N12E_C1
DSL-N12HP
DSL-N12U
DSL-N12U B1
DSL-N12U D1
DSL-N12U_C1
DSL-N14U
DSL-N14U B1
DSL-N16
DSL-N16U
DSL-N17U
DSL-N55U D1
DSL-N55U_C1
4G-AC68U
RT-AC65U
RT-AC85U
```

**general advice:**
```**10/18/2017 Security advisory for the vulnerabilities of WPA2 protocol ∇**
ASUS is aware of the recent WPA2 vulnerability issue. We take your security and privacy seriously and are currently working towards a full solution as quickly as possible. In the meantime, we want to help clarify the severity of the potential threat, and let our valued customers know the appropriate steps to take in order to avoid or lessen the threat of being compromised. 

Your devices are only vulnerable if an attacker is in physical proximity to your wireless network and is able to gain access to it. This exploit cannot steal your banking information, passwords, or other data on a secured connection that utilizes proper end-to-end encryption. However, an attacker could capture and read this information on an unsecured connection via an exploited WiFi network. Depending on the network configuration, it is also possible for the attacker to redirect network traffic, send invalid data to devices or even inject malware into the network. 

We are feverishly working with chipset suppliers to resolve this vulnerability and will release patched firmware for affected routers in the near future. Before this patched firmware is released, here are a few cautions all users should take: 

(1) Avoid public Wi-Fi and Hotspots until the routers and your devices are updated. Use cellular network connections if possible. 
(2) Only connect to secured services that you trust or have been verified. Web pages that use HTTPS or another secure connection will include HTTPS in the URL. If the connection is secured using TLS 1.2 your activities with that service is safe for now. 
(3) Keep your operating system and antivirus software up-to-date. Microsoft recently updated Windows to fix this exploit on their latest operating systems. Google and Apple are following suit shortly. 
(4) When in doubt, be safe and use your cellular network or a wired connection (Ethernet) to access the internet. This exploit only affects 802.11 traffic between a Wi-Fi router and a connected device on an exploited WiFi connection.
```

unsynaps, 2017-11-28:
@kernelOfTruth

It is because it only really effects devices acting as a client of some sort. So when it comes to base stations it only really effects them when they are acting as a bridge or range extender.

slavad, 2017-12-13:
>SDK update for AC66U and N66U were already merged with commit
@RMerl As far as understand this firmware is used for all supported models, right?

RMerl, 2017-12-13:
KRACK fixes are model specific.  380.69 only contains an updated wireless driver + related components for these two models, so they are the only models with a KRACK fix in this release.

Asus's fixes for other models are not compatible with the 380.xx codebase, so their fixes will only be available for the 382.xx releases.  RT-AC88U and RT-AC3100 were already fixed with 382.1_2.

slavad, 2017-12-14:
@RMerl got it. Current firmware for RT-AC68U is 3.0.0.4.**382**_18547-g91c580f, so as far as I understand this model will be fixed under 382.xx release. Am I right?