Asuswrt-Merlin/asuswrt-merlin.legacy #1517: IPv6: udhcpc/odhcp6c hard coded to request address, breaks DHCP-PD for some ISPs

xeth137, 2017-10-31:
My ISP (Wave G) uses SLAAC for IPv6 address assignment and DHCP-PD for prefix assignment.  If a DHCPv6 solicit is sent upstream with both an address request (IA_NA) and a prefix request (IA_PD), the server responds with status "NOADDRS-AVAIL", presumably because IA_NA is not configured.  In other words, DHCP-PD does not work for my ISP if address requests are sent in same solicit.

Here's a sample of the problematic behavior over the wire:

`Frame 12: 134 bytes on wire (1072 bits), 134 bytes captured (1072 bits)
Ethernet II, Src: IntelCor_XXXX, Dst: IPv6mcast_01:00:02 (33:33:00:01:00:02)
Internet Protocol Version 6, Src: XXXX, Dst: ff02::1:2
User Datagram Protocol, Src Port: 546, Dst Port: 547
DHCPv6
    Message type: Solicit (1)
    Transaction ID: 0x5f7860
    Elapsed time
        Option: Elapsed time (8)
        Length: 2
        Value: 02f4
        Elapsed time: 7560ms
    Option Request
        Option: Option Request (6)
        Length: 8
        Value: 0017001800520053
        Requested Option code: DNS recursive name server (23)
        Requested Option code: Domain Search List (24)
        Requested Option code: SOL_MAX_RT (82)
        Requested Option code: INF_MAX_RT (83)
    Client Identifier
        Option: Client Identifier (1)
        Length: 10
        Value: XXXX
        DUID: XXXX
        DUID Type: link-layer address (3)
        Hardware type: Ethernet (1)
        Link-layer address: XXXX
    Reconfigure Accept
        Option: Reconfigure Accept (20)
        Length: 0
    Identity Association for Non-temporary Address
        Option: Identity Association for Non-temporary Address (3)
        Length: 12
        Value: 000000010000000000000000
        IAID: 00000001
        T1: 0
        T2: 0
    Identity Association for Prefix Delegation
        Option: Identity Association for Prefix Delegation (25)
        Length: 12
        Value: 000110f20000000000000000
        IAID: 000110f2
        T1: 0
        T2: 0

Frame 13: 113 bytes on wire (904 bits), 113 bytes captured (904 bits)
Ethernet II, Src: Cisco_XXXX, Dst: IntelCor_XXXX
Internet Protocol Version 6, Src: XXXX, Dst: XXXX
User Datagram Protocol, Src Port: 547, Dst Port: 546
DHCPv6
    Message type: Advertise (2)
    Transaction ID: 0x5f7860
    Server Identifier
    Client Identifier
    Status code
        Option: Status code (13)
        Length: 15
        Value: 00024e4f41444452532d415641494c
        Status Code: NoAddrAvail (2)
        Status Message: NOADDRS-AVAIL
`

Looking through the code in udhcpc.c I see that it launches odhcp6c with the hardcoded option "-N try" which means it always sends an address request.  In my use case, this needs to be changed to "-N none" to avoid the issue.

My current workaround for this is use the wan-start user script to kill the odhcp6c instance started by udhcpc and start my own.  It would be nice if I can override the argument with an NVRAM flag or custom conf instead.

**TL/DR**: Need a way to configure odhcp6c to NOT send an address request (i.e. "-N none" instead of "-N try"), because it breaks IPv6 on some ISPs.  Please consider implementing this in 382 if not 380.

Firmware version 380.68_4, Model RT-AC56U

xeth137, 2017-10-31:
Here's a wire capture of the current (problematic) behavior:

[wireshark.txt](https://github.com/RMerl/asuswrt-merlin/files/1432040/wireshark.txt)

RMerl, 2017-10-31:
Can you provide a longer trace that would include the full dhcpv6 exchange?  I'll have someone take a look at them (since I don't know much about IPv6 myself, and can't test it either).

It's possible that the ISP isn't properly following RFCs, since -N try should normally be able to fallback if the request failed.

xeth137, 2017-10-31:
Thanks for the quick reply.  It looks like the fallback code only handles noaddrsavail in a REPLY, whereas in my case that flag is in an ADVERTISE (afaik it goes solicit -> advertise -> request -> reply).

I don't know much about IPv6 either so I'm not sure what the RFC behavior should be.  I'll try to post a full trace later today, but basically it's going to be the solicit/advertise pair I posted repeated over and over.

xeth137, 2017-11-01:
Here's a full trace of udp port 546/547 during bootup:

[vlan2_dhcpv6_anon.zip](https://github.com/RMerl/asuswrt-merlin/files/1432683/vlan2_dhcpv6_anon.zip)

I also dug through the IPv6 RFCs and it appears the server does not follow RFC 7550 (which was an update written specifically to address this issue), however that RFC also says clients should handle this legacy behavior for backwards compatibility.

https://tools.ietf.org/html/rfc7550#section-4.1

RMerl, 2017-11-01:
Can you test using this version?

http://www.mediafire.com/file/9cerclmabtpcfke/odhcp6c-test1.zip

Someone sent me a patch that might help make it more robust in such scenarios.  Test it by running it with "-N try".

My contact is currently busy with other things so he won't have the time to look at the trace for a few days.

RMerl, 2017-11-01:
Ok, try this test2 build now:

http://www.mediafire.com/file/gqc49k8snik1f5q/odhcp6c-test2.zip

Test it by running it with "-N try" to see if it can properly fallback.

xeth137, 2017-11-01:
running

```
odhcp6c -fve -R -s /tmp/dhcp6c -N try -c <SNIP> -PF 0:110f2 -r23 -r24 -r82 -r83 -k vlan2
```
results in syslog output of "invalid argument: 'F'"

...Taking away the F argument and everything works as expected.

```
odhcp6c -fve -R -s /tmp/dhcp6c -N try -c <SNIP> -PF 0:110f2 -r23 -r24 -r82 -r83 -k vlan2
```

It lo

But take away the F argument and everything works as expected.

```
odhcp6c -fve -R -s /tmp/dhcp6c -N try -c <SNIP> -PF 0:110f2 -r23 -r24 -r82 -r83 -k vlan2
```

Not using -F with the unpatched binary also fixes the issue.  It looks like force prefix flag prevents the -N try fallback code from working properly.

themiron, 2017-11-02:
@xeth137 you're using wrong -F & -P options order, -FP is valid.

xeth137, 2017-11-02:
My bad, using -FP with -N try does work with the patch.

RMerl, 2017-11-02:
Thanks for testing.  I`ll keep the issue open for now, please confirm that it`s still working correctly after a day or two.

@themiron : I plan on merging the updated odhcp6c (not just this patch) with 380.69 and 382.2.

themiron, 2017-11-02:
@RMerl great to know, thanks

themiron, 2017-11-02:
@xeth137 yep, thanks for testing, please make sure it correctly renews the prefixes after a while (not sure what's your lease time)

xeth137, 2017-11-02:
t1 is 900 and t2 is 1440, and it's been up for 6 hours now, looking good

xeth137, 2017-11-04:
Uptime of 47 hours with no issues.  Thanks to everyone for the fix!

closed 2017-11-04 by RMerl