Asuswrt-Merlin/asuswrt-merlin.legacy #1519: run entware-setup.sh get error

bigcat133, 2017-11-04:
### Package: <name>

For existing package:

update new firmware about meilin asusuwrt 382 beta3

For new package to be added:

i get error inf at run entware-setup.sh, error info as under:
`
--2017-11-04 16:38:10--  http://pkg.entware.net/binaries/armv7/installer/libc-2.23.so
Resolving pkg.entware.net... 81.4.123.217
Connecting to pkg.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1227404 (1.2M) [application/octet-stream]
Saving to: '/opt/lib/libc-2.23.so'

/opt/lib/libc-2.23.so              100%[=============================================================>]   1.17M   279KB/s    in 4.4s    

2017-11-04 16:38:15 (270 KB/s) - '/opt/lib/libc-2.23.so' saved [1227404/1227404]

/opt/lib/libgcc_s.so.1: No such file or directory
Info: Basic packages installation...
Collected errors:
 * opkg_conf_load: Couldn't stat /opt/etc/opkg.conf: No such file or directory.
Collected errors:
 * opkg_conf_load: Couldn't stat /opt/etc/opkg.conf: No such file or directory.
/opt/usr/lib/locale/locale-archive: No such file or directory
Info: Congratulations!
Info: If there are no errors above then Entware-ng was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware-ng services to start
Info: Found a Bug? Please report at https://github.com/Entware-ng/Entware-ng/issues
`

### Platform:
my router is asus ac-86u , the run script entware-setup.sh is in the meilin asusuwrt 382 beta3 version

RMerl, 2017-11-12:
Works for me.  Check your USB disk, remove any existing Entware folder and retry.

```
ASUSWRT-Merlin RT-AC86U 382.1-beta4-gd8be69c Thu Nov  9 04:14:26 UTC 2017
admin@Stargate86:/tmp/home/root# entware-setup.sh 
 Info:  This script will guide you through the Entware installation.
 Info:  Script modifies "entware" folder only on the chosen drive,
 Info:  no other data will be changed. Existing installation will be
 Info:  replaced with this one. Also some start scripts will be installed,
 Info:  the old ones will be saved on Entware partition with name
 Info:  like /tmp/mnt/sda1/jffs_scripts_backup.tgz

 Info:  Looking for available partitions...
[1] --> /tmp/mnt/sda1
 =>  Please enter partition number or 0 to exit
[0-1]: 1
 Info:  /tmp/mnt/sda1 selected.

 Info:  Creating /tmp/mnt/sda1/entware folder...
 * Warning:  Deleting old /tmp/opt symlink...
 Info:  Creating /tmp/opt symlink...
 Info:  Creating /jffs scripts backup...
tar: removing leading '/' from member names
 Info:  Modifying start scripts...
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
--2017-11-11 22:46:04--  http://pkg.entware.net/binaries/armv7/installer/opkg
Resolving pkg.entware.net... 81.4.123.217
Connecting to pkg.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 127488 (124K) [application/octet-stream]
Saving to: '/opt/bin/opkg'

/opt/bin/opkg                         100%[=========================================================================>] 124.50K   284KB/s    in 0.4s    

2017-11-11 22:46:04 (284 KB/s) - '/opt/bin/opkg' saved [127488/127488]

--2017-11-11 22:46:04--  http://pkg.entware.net/binaries/armv7/installer/opkg.conf
Resolving pkg.entware.net... 81.4.123.217
Connecting to pkg.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 142 [application/octet-stream]
Saving to: '/opt/etc/opkg.conf'

/opt/etc/opkg.conf                    100%[=========================================================================>]     142  --.-KB/s    in 0s      

2017-11-11 22:46:05 (19.2 MB/s) - '/opt/etc/opkg.conf' saved [142/142]

--2017-11-11 22:46:05--  http://pkg.entware.net/binaries/armv7/installer/ld-2.23.so
Resolving pkg.entware.net... 81.4.123.217
Connecting to pkg.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 142752 (139K) [application/octet-stream]
Saving to: '/opt/lib/ld-2.23.so'

/opt/lib/ld-2.23.so                   100%[=========================================================================>] 139.41K   271KB/s    in 0.5s    

2017-11-11 22:46:06 (271 KB/s) - '/opt/lib/ld-2.23.so' saved [142752/142752]

--2017-11-11 22:46:06--  http://pkg.entware.net/binaries/armv7/installer/libc-2.23.so
Resolving pkg.entware.net... 81.4.123.217
Connecting to pkg.entware.net|81.4.123.217|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1227404 (1.2M) [application/octet-stream]
Saving to: '/opt/lib/libc-2.23.so'

/opt/lib/libc-2.23.so                 100%[=========================================================================>]   1.17M   636KB/s    in 1.9s    

2017-11-11 22:46:08 (636 KB/s) - '/opt/lib/libc-2.23.so' saved [1227404/1227404]
[... 72 lines cut ...]
Info: Congratulations!
Info: If there are no errors above then Entware-ng was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware-ng services to start
Info: Found a Bug? Please report at https://github.com/Entware-ng/Entware-ng/issues
admin@Stargate86:/tmp/home/root# opkg update
Downloading http://pkg.entware.net/binaries/armv7/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/packages
admin@Stargate86:/tmp/home/root# opkg install zork
Installing zork (1.0-20140806-1) to root...
Downloading http://pkg.entware.net/binaries/armv7/zork_1.0-20140806-1_armv7soft.ipk
Configuring zork.
admin@Stargate86:/tmp/home/root# dungeon
Welcome to Dungeon.			This version created 11-MAR-91.
You are in an open field west of a big white house with a boarded
front door.
There is a small mailbox here.
>quit
Your score would be 0 [total of 585 points], in 1 move.
This gives you the rank of Beginner.
Do you wish to leave the game?
y
The game is over.
admin@Stargate86:/tmp/home/root# 
```

bigcat133, 2017-11-16:
thanks @RMerl, i success run the script when update firmware to 382.1

closed 2017-11-16 by RMerl