Asuswrt-Merlin/asuswrt-merlin.legacy #1523: 【AC86U 382.1 beta 4】Firewall cause NAT not work.

iMocW, 2017-11-10:
Here's iptables-save output:

[firewalldisabled.txt](https://github.com/RMerl/asuswrt-merlin/files/1461986/firewalldisabled.txt)
[firewallenabled.txt](https://github.com/RMerl/asuswrt-merlin/files/1461987/firewallenabled.txt)

When firewall is on, subnet devices except router itself can not access the internet.
Just turn it off to make link up.

iMocW, 2017-11-10:
Lack of `iptables -t filter -I FORWARD ! -i ppp0 -j ACCEPT`, in other words: FORWARD chain in filter table causes this problem.

iMocW, 2017-11-11:
This also causes port forwarding stops working !!!

RMerl, 2017-11-12:
Remove any customization you've made.  There are a few hundred users out there running this firmware without any problem with their WAN connection, so something on your setup is causing it to break.

zeit200, 2017-11-19:
Got the same problem after enabling openvpn server
[iptables.txt](https://github.com/RMerl/asuswrt-merlin/files/1485179/iptables.txt)
[server1-fw.sh.txt](https://github.com/RMerl/asuswrt-merlin/files/1485180/server1-fw.sh.txt)
[config.ovpn.txt](https://github.com/RMerl/asuswrt-merlin/files/1485181/config.ovpn.txt)

RMerl, 2017-11-19:
Already fixed with https://github.com/RMerl/asuswrt-merlin.382/commit/57502aeede08940fdfb704e32b47aa1ba9813d44 .

closed 2017-11-19 by RMerl

iMocW, 2017-11-25:
@RMerl No. Still needs `-A FORWARD -i br0 -o ppp0 -j ACCEPT` to make NAT work right. [382.1]

iMocW, 2017-11-25:
@RMerl  And fake LAN TTL will cause some of LAN side devices speed near to 0Mbps but/when ping is right.