Asuswrt-Merlin/asuswrt-merlin.legacy #1524: DNSMASQ can not start. 382.1_beta4,AC86U

iMocW, 2017-11-12:
there are 32000+ lines(server=domain/dnsserver) dnsmasq.conf.add, dnsmasq exits due to "fail to allocate memory".

never happened on AC68U with 380.68.

RMerl, 2017-11-12:
You will need to rethink your configuration, as 32000+ lines is way excessive.  If the router says it doesn't have enough memory, then there's nothing I can do about it.  Firmware 382 uses more RAM due to additional features and the new Trend Micro engine, and the RT-AC86U also reserves a large portion of the RAM for the Broadcom components.

closed 2017-11-12 by RMerl

iMocW, 2017-11-15:
I don't think that's the case.
I can not run latest(171113) kcptun client (arm5/6/7) when it had 150MB+ free memory just now:
```
fatal error: runtime: out of memory

runtime stack:
runtime.throw(0x204726, 0x16)
        /usr/local/go/src/runtime/panic.go:605 +0x70 fp=0xfff7d428 sp=0xfff7d41c pc=0x3a838
runtime.sysMap(0xf780000, 0x1080000, 0x1ed01, 0x32e790)
        /usr/local/go/src/runtime/mem_linux.go:227 +0x104 fp=0xfff7d44c sp=0xfff7d428 pc=0x2457c
runtime.(*mheap).mapBits(0x321da8, 0x10800000)
Mem: 301868K used, 138552K free, 0K shrd, 864K buff, 37776K cached
```

iMocW, 2017-11-15:
This client uses only about 1MB memory when startup.(Tested on my AC68U)