Asuswrt-Merlin/asuswrt-merlin.legacy #1529: set iptables return "No match by name"

bigcat133, 2017-11-16:
i run the command as that:
`ipset -N gfwlist iphash
iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
`
get the error info as:
`iptables: No chain/target/match by that name.`

but i run same command at wndr4300 is ok
wndr 4300: Linux OpenWrt 3.18.20 #1 Fri Sep 4 22:36:47 CEST 2015 mips GNU/Linux
iptables v1.4.21

platform:
asus ac-86u run firmware 382.1
iptables v1.4.15

thank for all

RMerl, 2017-11-16:
Manually modprobe the kernel module first:

```
modprobe xt_set
```

382.2 added the module to the modprobe alias list so it will be automatically loaded.

bigcat133, 2017-11-16:
thanks, that ok

closed 2017-11-16 by RMerl