Asuswrt-Merlin/asuswrt-merlin.legacy #1536: Wrong time in system log BUT only for some applications

jbaker6953, 2017-11-24:
Found this in my RT-AC88U Merlin 380.68_4 system log this morning (note the times for dnscrypt-proxy are an hour ahead of everything else:

`Nov 24 02:58:04 dnscrypt-proxy[1233]: Refetching server certificates`
`Nov 24 02:58:04 dnscrypt-proxy[1233]: Server certificate with serial '0001' received`
`Nov 24 02:58:04 dnscrypt-proxy[1233]: This certificate is valid`
`Nov 24 02:58:04 dnscrypt-proxy[1233]: Chosen certificate #808464433 is valid from [2017-04-12] to [2018-04-12]`
`Nov 24 02:58:04 dnscrypt-proxy[1233]: The key rotation period for this server may exceed the recommended value. This is bad for forward secrecy.`
`Nov 24 02:58:04 dnscrypt-proxy[1233]: Server key fingerprint is 08A8:0048:947A:E4BB:E509:EC71:2A01:4E35:36D9:4E3D:880F:118B:F484:7FE4:6F98:0F44`
`Nov 24 02:00:14 rc_service: rc 31508:notify_rc restart_wrs`
`Nov 24 02:03:29 openvpn[1561]: android/192.168.229.23 TLS: soft reset sec=0 bytes=38969/-1 pkts=454/0`
`Nov 24 02:03:29 openvpn[1561]: android/192.168.229.23 TLS ERROR: local/remote key IDs out of sync (1/2) ID:  [key#0 state=S_START id=1 sid=9b45e86b 85006920] [key#1 state=S_NORMAL_OP id=0 sid=9b45e86b 85006920] [key#2 state=S_UNDEF id=0 sid=00000000 00000000]`
`Nov 24 02:03:29 openvpn[1561]: android/192.168.229.23 Fatal TLS error (check_tls_errors_co), restarting`
`Nov 24 02:03:29 openvpn[1561]: android/192.168.229.23 SIGUSR1[soft,tls-error] received, client-instance restarting`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: Refetching server certificates`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: Server certificate with serial #1496061161 received`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: This certificate is valid`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: Chosen certificate #1496061161 is valid from [2017-05-29] to [2027-05-27]`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: The key rotation period for this server may exceed the recommended value. This is bad for forward secrecy.`
`Nov 24 03:04:21 dnscrypt-proxy[1242]: Server key fingerprint is E737:6400:D646:0720:7D9D:29AB:A4C9:070C:4546:CEF7:0CFE:D62F:41E9:FEAA:C58F:6376`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: Refetching server certificates`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: Server certificate with serial #1501476487 received`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: This certificate is valid`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: Chosen certificate #1501476487 is valid from [2017-07-31] to [2027-07-29]`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: The key rotation period for this server may exceed the recommended value. This is bad for forward secrecy.`
`Nov 24 03:04:27 dnscrypt-proxy[1256]: Server key fingerprint is B161:86A9:B925:8999:2CB4:6E43:3979:A018:5AAD:623A:33B0:CEB9:40A7:27B4:38EC:EC3D`
`Nov 24 02:06:01 openvpn[1561]: TCP connection established with [AF_INET6]::ffff:192.168.229.23:41551`

RMerl, 2017-11-24:
Reboot your routers, the DST change didn't get applied to services that were already running.

jbaker6953, 2017-11-24:
Router was rebooted yesterday:

`Nov 23 09:44:29 kernel: DROP IN=eth0 OUT= MAC=10:7b:44:ad:b1:e8:dc:38:e1:d1:89:c4:08:00 SRC=210.194.61.23 DST=47.149.32.39 LEN=40 TOS=0x00 PREC=0x00 TTL=46 ID=11587 PROTO=TCP SPT=11757 DPT=23 SEQ=798302247 ACK=0 WINDOW=13814 RES=0x00 SYN URGP=0 `
`Nov 23 09:45:36 kernel: DROP IN=eth0 OUT= MAC=10:7b:44:ad:b1:e8:dc:38:e1:d1:89:c4:08:00 SRC=191.101.167.7 DST=47.149.32.39 LEN=40 TOS=0x00 PREC=0x00 TTL=243 ID=6323 PROTO=TCP SPT=54793 DPT=4146 SEQ=92088486 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 `
`Nov 23 09:45:45 kernel: DROP IN=eth0 OUT= MAC=10:7b:44:ad:b1:e8:dc:38:e1:d1:89:c4:08:00 SRC=190.174.198.213 DST=47.149.32.39 LEN=40 TOS=0x00 PREC=0x00 TTL=52 ID=1112 PROTO=TCP SPT=34597 DPT=23 SEQ=798302247 ACK=0 WINDOW=48922 RES=0x00 SYN URGP=0 `
`Jul 31 17:00:16 syslogd started: BusyBox v1.25.1`
`Jul 31 17:00:16 kernel: klogd started: BusyBox v1.25.1 (2017-10-04 15:03:05 EDT)`
`Jul 31 17:00:16 kernel: Linux version 2.6.36.4brcmarm (merlin@ubuntu-dev) (gcc version 4.5.3 (Buildroot 2012.02) ) #1 SMP PREEMPT Wed Oct 4 15:12:02 EDT 2017`
`Jul 31 17:00:16 kernel: CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c53c7f`
`Jul 31 17:00:16 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache`
`Jul 31 17:00:16 kernel: Machine: Northstar Prototype`

jbaker6953, 2017-11-24:
Could it be an issue with the DST or TZ propagating to the Entware environment?

jbaker6953, 2017-11-24:
Trying `ln -s /etc/TZ /mnt/sda1/entware/etc/TZ`

jbaker6953, 2017-11-24:
Nope. That causes log entries to be in GMT.

jbaker6953, 2017-11-24:
Instructions at https://github.com/Entware-ng/Entware-ng/wiki/Using-time-zones also result in GMT times in the log

jbaker6953, 2017-11-24:
OK. Jeez. `ln -s /etc/TZ /mnt/sda1/entware/etc/TZ` with router reboot fixed it. It was not enough to restart the affected services.

closed 2017-11-25 by RMerl