Asuswrt-Merlin/asuswrt-merlin.legacy #1546: Feature Request: Add hostname as an arg to the ddns-start script

mike-pt, 2017-11-30:
Right now if we setup custom ddns, the router will pass the IP in the 1st argument, but this page asks for a domain name that's not being used (except maybe to show it in Internet Status).

If we could pass this as a 2nd arg to ddns-start it would enable us to use that host name in a script easier.

I noticed the need of this cause If you have more than one connection/location using asus routers and can't use assuscomm, we would need to setup the script in every router but in all the "name" is hardcoded in the script, if we could simple re-used the hostname set in this page it enables easier distribution of the script to different routers and allows an admin to update the hostname if needed.

mike-pt, 2017-11-30:
Looking at services.c it seems we already use `ddns_hostname_x` in other services and could pass it to ddns-start, *seems* like an easy change but I'm no c expert.

But reading this hinted me to the fact that in the mean time we can use `nvram get ddns_hostname_x` in the script it self.

RMerl, 2017-12-08:
The FQDN is used for places like OpenVPN where it wants to use one.  I don't pass it as an argument to the script because it's a static parameter, which can be retrieved from nvram - I mostly pass only dynamic parameters to those scripts.

closed 2017-12-08 by RMerl