Asuswrt-Merlin/asuswrt-merlin.legacy #1561: Scheduled reboot not working RT68U 380.69

omegadcuk1965, 2017-12-18:
Scheduled rebooting at say 4am in the morning appears to leave WIFI no working, only seems to come back to life once you physically power cycle the router.  

I did quite a few tests on scheduled reboot between 11am and 2pm and it was working fine? weird.  Saw some reports in an old issue thread (closed) that it can fail anywhere between 10% and 90% of the time, so maybe it was just lucky.

kiwilad2, 2018-01-11:
Hi I have an RT-AC87U with the same issue.  the Scheduled reboot causes the WiFi to stop working and you need to power cycle the unit to get it to recover.  I've gone back to 380-68-2 to get around this issue.
Does anyone have an update?

thefad3done, 2018-01-18:
I have the same issues with my RT-AC87U - since turning off scheduled reboot, I've not had any issues with my WiFi / router stalling.

unsynaps, 2018-01-18:
87 here as well but have had zero issues with scheduled reboots. IIRC it’s set to do it once a week.

thefad3done, 2018-01-22:
I have since determined that the reboot scheduler wasn't the issue - seems I have issues running the latest firmware so have had to go back a version (which has solved my issues)