Asuswrt-Merlin/asuswrt-merlin.legacy #1568: Guest network feature broken when using DNS servers on intranet

wrffrz, 2017-12-24:
I'm using an RT-AC5300 and have a pair of Raspberry Pis running Pi-Hole on my intranet configured as DNS servers 1 & 2, which works great for network-wide ad-blocking.

But when a Guest network has been enabled and set to block access to the intranet, the router still tells guest clients to use my local DNS servers which are unreachable.

My only option is allow access to my local network from the Guest network which defeats the only entire purpose of the feature.

haa-zee, 2017-12-28:
A workaround: create a new subnet and put your DNS servers to there (you could add multiple IP addresses to the same interface, so they can be both in local and "non-local" subnets at the same time).
The router will allow guests to access that IPs :( but not to the own subnet.

RMerl, 2017-12-28:
The router is designed expecting to act as the DNS on your LAN.  You're trying to use a non-standard setup, which is bound to create issues - your setup will also break LAN-side name resolution for instance.

A proper setup would be to keep using the router as the DNS provided through DHCP to your clients, but configure the WAN interface to use your Pi-Hole IPs as DNS.  That way, you will have properly working LAN name resolution, Guest clients will be able to get working name resolution (including the security provided by your Pi-Hole).  The router will act as a caching nameserver, like it's meant to be, with the Pi-Hole acting as the recursive resolvers.

closed 2017-12-28 by RMerl

wrffrz, 2017-12-29:
Actually, my LAN-side name resolution was working great with this setup.  The way it was configured was--

router is 192.168.1.1
pi-hole1 & pi-hole2 are 192.168.1.10 and .11 and set to use router as upstream DNS

router settings:
WAN->DNS Server->Automatic
LAN->DHCP->DNS Servers 1 & 2-> 192.168.1.10 and 192.168.1.10

I've switched over to the setup you suggested, but had to power-cycle the router to get it to come back up after applying the settings.  It hung and wasn't responding to pings or broadcasting it's SSID for >15 minutes before I pulled the power.  I'll test further and see how well this works.

If LAN->DHCP Server->DNS Server 1 & 2 settings should not be used, they should probably be removed from the configuration interface.