Asuswrt-Merlin/asuswrt-merlin.legacy #1569: Imported .ovpn containing two certificates causes http server to die requiring factory reset

fortran77, 2017-12-25:
Importing the attached .ovpn configuration file into an ASUS RT-N66R router
(firmware release: 380.69) causes its embedded http server to die, and to 
continue to repeatedly die any time any item in the "Select client instance"
pulldown list on the VPN Clients page is selected.

Once this has happened, the only fix is to factory-reset the router.

I also tested with a number of earlier firmware versions and the behavior was
the same.

The supplied .ovpn configuration is not claimed to be valid (although it 
probably is, as I hand-crafted it from one of the configurations available on 
the website of Private Internet Acces by embedding their certificate authority
certificate twice into that configuration).

The feature of the configuration that triggers the problem is the certificate 
being included twice. The hand-crafted example supplied here includes the same
certificate twice. In practice the problem would normally be caused by a VPN
provider that uses two different certificates that are chained together.

[example-that-triggers-bug.ovpn.txt](https://github.com/RMerl/asuswrt-merlin/files/1585805/example-that-triggers-bug.ovpn.txt)

RMerl, 2017-12-28:
I'll take a look at the cause of the crash, but note that such a certificate is too large to fit in the 3499 chars limit supported by the firmware, so it will never work properly.

RMerl, 2017-12-28:
Crash reproduced under 380.69, but not with 382.2 (which has stricter nvram validation measures in place).

RMerl, 2017-12-28:
```
(gdb) backtrace
#0  0x4060f014 in strcmp () from /lib/libc.so.0
#1  0x0001034c in call (stream=0xc96ad, func=0x45544143 <error: Cannot access memory at address 0x45544143>) at ej.c:103
#2  process_asp (f=0xc96ad, e=0xbed39e40 "%>\n<% vpn_crt_server(); %>\n\n", s=0xbed39e2d " vpn_crt_client") at ej.c:128
#3  do_ej (path=<optimized out>, stream=0xc96ad) at ej.c:365
#4  0x0000eb78 in handle_request () at httpd.c:1362
#5  0x0000f734 in main (argc=<optimized out>, argv=<optimized out>) at httpd.c:2237
```

closed 2017-12-28 by RMerl

fortran77, 2017-12-28:
That was a very quick fix! Thanks.

commit 30b06536 (2017-12-28): shared: fix buffer overrun in get_parsed_crt() if provided with a key/cert longer than the supported limit (closes #1569)