Asuswrt-Merlin/asuswrt-merlin.legacy #1574: [382.2_Beta 1] iptables.filter.FORWARD chain is broken again.

iMocW, 2017-12-31:
Lack of 
-i br0 -o ppp0 -j ACCEPT 
-i br0 -o tun11 -j ACCEPT 
First one causes WAN break. 
Second one causes VPN break. 

PS1. Just simply use -i br0 -j ACCEPT to fix it?
PS2. SSH brute force protection ON.(not sure if it's functional)

iMocW, 2017-12-31:
And lack of DNAT trace ACCEPT causes access from WAN break.
"...-m conntrack --ctstate DNAT..."

iMocW, 2017-12-31:
And can be simply fixed by Applying again settings on page Advanced_System_Content.asp(no settings changed).

RMerl, 2018-01-01:
Neither of these rules are necessary, since the last rule in the chain covers -i br0 -o *.

```
 0     0 ACCEPT     all  --  br0    *       0.0.0.0/0            0.0.0.0/0  
```

There also already a DNAT rule in there:

```
0     0 ACCEPT     all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate DNAT
```

iMocW, 2018-01-01:
I know. The main problem is there is a chance(always for me) these rules are disappearing. Need to re-apply the settings on the page which I mentioned before.

iMocW, 2018-01-01:
After boot/after stop and start(not restart) the VPN client/etc..

iMocW, 2018-01-02:
Beta 2, issue still.

iMocW, 2018-01-02:
And now can not be fixed by re-apply settings on page Advanced_System_Content.asp