Asuswrt-Merlin/asuswrt-merlin.legacy #1577: Bandwidth limiting cannot be set to 0 (default)

perfecthumaninterface, 2018-01-06:
When configuring QoS, attempting to set the bandwidth limiting values back to 0 (which is stated to be the default that indicates bandwidth is unlimited), an error is thrown stating that it cannot be set to 0.

[image]

This is firmware 380.69 on an RT-AC87R.

unsynaps, 2018-01-06:
Well first off those are not the bandwidth limiting values. Those values are to tell the QoS "engine" what your maximum bandwidth is. AKA the most the ISP will give you based on your rate plan.

Aside from that yeah. Should be able to put 0 (zero) in there.

perfecthumaninterface, 2018-01-07:
Are you sure? It does have the effect of limiting the bandwidth. I also don't see why it would have an option for "unlimited" bandwidth if this setting was not intended to limit bandwidth.

ChriscomIT, 2018-01-07:
Just set the bandwidth setting to automatic and it should work.

unsynaps, 2018-02-03:
@perfecthumaninterface
For the system to calculate the percentages on its own (adaptive) it needs to know the upper end  to give the proper bandwidth per data type.

If you are told to give someone 60% water in their glass how do you calculate that without knowing how much the glass holds?

perfecthumaninterface, 2018-02-03:
It states right in the UI that "the default is 0, which means unlimited bandwidth." It was set to 0 before I changed it. So are you suggesting that it can't be set to 0 with the current settings (adaptive qos) but it can be with others?