Asuswrt-Merlin/asuswrt-merlin.legacy #1578: Looking for a better (more simplified) examples of OpenVPN redirect policies.

electricessence, 2018-01-09:
```No``` does what? 
```All``` does what?

What's the difference between policies if strict or not?

RMerl, 2018-01-09:
All documented in the Wiki:

https://github.com/RMerl/asuswrt-merlin/wiki/Policy-based-routing

closed 2018-01-09 by RMerl

electricessence, 2018-01-10:
Should have mentioned that I read that and that's why I opened this ticket.  :|
Sorry if I don't understand 100% what's written there, but I would hope to see a table that has a simple non-technical explanation for the options.
X: Routes all traffic through
Y: Routes none of the traffic.

I have a hard time discerning what each does by default and therefore subsequently what the risks are at using each.

RMerl, 2018-01-10:
Best you ask on the support forums at SNBForums then.  I can't really make the documentation any simpler than it currently is - the Wiki's goal isn't to explain the basics about networking and VPNs, that would be outside of the scope of this project.