Asuswrt-Merlin/asuswrt-merlin.legacy #1581: Can not get to webui thrue vpn, or might be from local also

Bardissimo, 2018-01-12:
Hi. Currently i can not ping from wan my AC-5300 380.68_2, but can reach webui.
When connected to VPN cannot access webui by local address. Tried disabling/enabling firewall - no luck. But when i enabled packet logging i have found that packets are dropped, if i`am trying to reach from VPN from TUN int, when accessing from TAP int (second vpn server) it does not even shows in log.

[block.txt](https://github.com/RMerl/asuswrt-merlin/files/1625273/block.txt)

but i can ping local ip and access server by ssh, when connected thrue vpn

bfikkert, 2018-01-12:
I'm not sure but,
Have you read this? Unther example, the last rule?

[https://github.com/RMerl/asuswrt-merlin/wiki/Policy-based-routing](url)

Bardissimo, 2018-01-13:
Well it was solved.
Few things were need to be done.
1) my eyes were soaped )) from lan connection to webui is going to port 80, not 8000 i`am using for access from WAN
2) power sycle solved issue with icmp requests from wan

closed 2018-02-04 by RMerl