Asuswrt-Merlin/asuswrt-merlin.legacy #1585: Changing nvram country code (RT-AC88U)

Blusy38, 2018-01-21:
Hello,
I just bought a router RT-AC88U from a Chinese site (Gearbest).
I installed Merlin 382.1_2.

The frequencies of the wifi 5 Ghz are limited to the Chinese frequencies 149 to 156 (U-NII-3) but I would like to use the European frequencies 36 to 48 (U-NII-1).

I can not change the country code, even in SSH.

The command :

`nvram show | grep "country_code \ | regulation_domain \ | reg_mode \ | ccode"`

Reference:

```
wl0_country_code=CN
pci/2/1/ccode=EU
1:ccode=CN
size: 64913 bytes (66159 left)
wl_country_code=EU
0:ccode=CN
wl0_reg_mode=off
wl1_reg_mode=off
pci/1/1/ccode=EU
wl1_country_code=CN
regulation_domain=EU
regulation_domain_5G=EU
wl_reg_mode=off

```
I can not change CN codes in EU, even with:

```
nvram set pci / 1/1 / ccode = EU
nvram set pci / 2/1 / ccode = EU
nvram set wl0_country_code = EU
nvram set wl1_country_code = EU
```

Do you know if it's possible or if it's a legal block? Do you have a solution ?
Thank you for your help.
Guillaume.

Marco-vW, 2018-01-21:
This might help: [https://www.snbforums.com/threads/rt-ac68u-change-region.14815/ ](https://www.snbforums.com/threads/rt-ac68u-change-region.14815/ )(you need to verify if the nvram variables are identical, as this thread concerns the RT-AC68U). 

For support [https://www.snbforums.com/forums/asuswrt-merlin.42/](https://www.snbforums.com/forums/asuswrt-merlin.42/) is the place to be.

Blusy38, 2018-01-21:
Thank you for your help but unfortunately it is no longer possible to change the wireless settings by changing nvram.
https://www.snbforums.com/threads/rt-n66u-merlin-cant-update-nvram-settings.33170/#post-266424

Marco-vW, 2018-01-21:
Sorry to hear that. Sounds like you've answered your own question then. If it's hardcoded in the bootloader, you're probably stuck with the settings of the country of origin :-/

newsera, 2018-01-28:
This worked for me. 

nvram set asuscfe0:ccode=AU
nvram set asuscfe1:ccode=AU
nvram set asuscfe0:regrev=0
nvram set asuscfe1:regrev=0
nvram set asuscfecommit=1
nvram set wl0_country_code=AU
nvram set wl0_country_rev=0
nvram set wl0_reg_mode=off
nvram set wl1_country_code=AU
nvram set wl1_country_rev=0
nvram set wl1_reg_mode=off
nvram set wl_txpower=250
nvram set wl0_txpower=250
nvram set wl1_txpower=250
nvram commit
reboot

using stock "US"; UNITED STATES

wl txpwr_target_max 19.0 19.0 19.0 (about 80mW)

2.4 Ghz channels: 1-11

5Ghz channels:36 40 44 48 149 153 157 161 165 

using "SG": SINGAPORE

wl txpwr_target_max 19.5 19.5 19.5 (about 90mW)

2.4 Ghz channels: 1-13

5Ghz channels: Same as Stock "US" ccode 36 40 44 48 149 153 157 161 165 

using "AU": AUSTRALIA

wl txpwr_target_max 23.5 23.5 23.5 (about 224mW)

2.5Ghz channels: 1-13

5Ghz Channels:36 40 44 48 52 56 60 64 100 104 108 112 116 132 136 140 144 149 153 157 161 165

using "#a":

wl txpwr_target_max 25.0 25.0 25.0 (about 316mW)

wl0_chlist=1 2 3 4 5 6 7 8 9 10 11 12 13 14

wl1_chlist=34 36 38 40 42 44 46 48 52 56 60 64 100 104 108 112 116 120 124 128 132 136 140 144 149 153 157 161 165

Blusy38, 2018-01-28:
Thank you for your help, but unfortunately it still does not work for me.
Did you do it on a RT-AC88U from China?
The Chinese version may be locked.

```
wl0_country_code=CN
1:ccode=CN
0:ccode=CN
wl1_country_code=CN
```

```
nvram show | grep "country_code\|regulation_domain\|reg_mode\|ccode"

wl0_country_code=CN
pci/2/1/ccode=AU
1:ccode=CN
size: 64905 bytes (66167 left)
wl_country_code=AU
0:ccode=CN
wl0_reg_mode=off
wl1_reg_mode=off
asuscfe1:ccode=AU
w10_country_code=EU
pci/1/1/ccode=AU
wl1_country_code=CN
regulation_domain=AU
regulation_domain_5G=AU
wl_reg_mode=h
asuscfe0:ccode=AU
```
Did you do anything else before you could make the change?
Thank you.

RMerl, 2018-02-04:
Changing region code is not supported.  It gets overwritten by the bootloader values.

closed 2018-02-04 by RMerl