Asuswrt-Merlin/asuswrt-merlin.legacy #1586: WiFi slow/dropping out

thefad3done, 2018-01-22:
On the latest Legacy firmware (380.69) - I have experienced 5ghz band WiFi issues galore which could only be resolved (for a small period of time) with a powercycle.

Issues that occurred with the 5ghz band:
- Slowness (speeds of 3-5mb/s)
- Signal dropping out
- Failure to connect for some devices
- 5ghz taking much longer than usual to start up

Initially I thought it was the reboot scheduler as per another user with the same router that was having some issues. After turning it off, things seemed like they improved but alas they did not.

Some troubleshooting I performed:
- Complete wipe of the firmware
- reinstalled the firmware
- added devices 1 by 1 to see if any of them were causing issues (they were not)

Unfortunately I was forced to back a version to 380.68_4 which proved to be stable for me (and has been for the last few days).

Router Model: RT-AC87U

sock5puppet, 2018-01-25:
Running into the exact same problems with a brand new, clean 5300 series. I am likely going to have to revert to factory firmware just to get the 5ghz stable. I don't see any errors in logs/configs. Maybe I'm not looking in the right place but so far I haven't found any specific reason for this.

Something that "appeared" to help a little was manually setting the control channel but I have yet to vet that out fully.

RMerl, 2018-02-04:
Wifi is outside of my control.

Also, the RT-AC87U 5 GHz firmware hasn't changed in nearly 2 years.

closed 2018-02-04 by RMerl